waveform80/picamera #185: GPIO input setting lost after calling picamera function

jeffboyce, 2015-01-18:
I am not sure whether this would be a question, a feature request, or a bug.  In summary, I was setting up a passive infrared (motion) sensor to activate my camera.  I used GPIO.setmode(GPIO.BOARD) in my sensor routine, then called another routine to set the camera parameters and capture a still image.  This was set in a simple loop as motion was detected, or not detected.  The first time through the loop a photo would be taken.  However, the second time through would produce the error RuntimeError: You must setup() the GPIO channel first.  A complete description of my issue is posted at http://www.raspberrypi.org/forums/viewtopic.php?f=32&t=96927.  The problem is that the picamera is using GPIO.setmode(GPIO.BCM) to do something with the LED (see def _init_led(self): in source code).  This has the effect of clearing the GPIO if it has previously been set with GPIO.BOARD.  It took a lot of searching and headscratching for me to figure this out.  Since the camera is not plugged into any of the GPIO pins, it is not apparent that there would be any reason it would use setmode for the GPIO pins.

I am not sure why the camera needs to interact with the GPIO pins if it doesn't connect to them, but is there a way to change this section of code so that it would not conflict with someone using GPIO.BOARD mode, or either mode?  Thanks.

Jeff

waveform80, 2015-01-21:
Unfortunately the camera's LED is only controllable when the GPIO library is in BCM mode (there's no mapping for pin 5 in BOARD mode). Hence, I've no choice but to switch the library into BCM mode when the led property is used. Ideally I'd like to have queried which mode the GPIO library was in and thrown an error if the led property is accessed when the library is in BOARD mode; unfortunately, the GPIO library provides no facility for querying which mode it's in so the best I can do is to arbitrarily switch it to BCM mode and hope things are fine.

That said, I have written the code tolerably conservatively: note that `_init_led` (which is what sets RPi.GPIO to BCM mode) isn't called until the first attempt to set the camera's led property. In other words, if the user is using the GPIO library in BOARD mode, and never touches the camera's led property then everything will be fine. The problem only arises if the user uses GPIO in BOARD mode and then attempts to set the camera's led property. It looks like your code (from the linked forum post) is attempting to set the led in the `SinglePhoto` function so I'm afraid your only solutions in this case are to either avoid setting camera.led or to use BCM mode instead.

waveform80, 2015-01-21:
I should add that there's a note about this in the [LED recipe](http://picamera.readthedocs.org/en/release-1.9/recipes1.html#controlling-the-led) in the docs, but admittedly not in the LED API reference - I should probably add one there.

jeffboyce, 2015-01-22:
Thanks for the explanation.  I did fix my problem by switching my routine to use BCM mode.  It looks like the only long term solution to this would be to ask for a feature request for the GPIO library to include a way to query what mode it is in (if that is even possible).  I think what added to my frustration in trying to track this down, was that I didn't know that it was the led property in the first place that was causing the issue.  It took someone much smarter than I to speculate that it might have been some other routine using the pins (hence trying different pins first), then read the code and find that the led property was using BMC mode, and hypothesize that might be my problem.  So while I had read that section of the docs about using the led, it didn't jump out to me that it would be my problem.  Probably because I didn't make the connection that the led is using a pin, when there is no physical connection of the camera to the pins.  I still don't have a complete understanding of how that works, but it is time to move on to other things.

closed 2015-01-22 by jeffboyce