waveform80/picamera #187: Producing a RPIRGBArray

LeeKnudtson13, 2015-01-26:
I'm a researcher with the University of Wisconsin-Milwaukee.  I'm currently working on a multispectral camera project for coastal monitoring.  I am utilizing the Raspberry Picam and python-picamera (using Python 3).  I'm currently trying to view a RPIRGBArray but the print function is not working how I expect it too.  I expect the script to print a 3D matrix representing pixel and color values.  When I run the script, I get a result that is depicted in the attachment.  Can you help me achieve a 3D RGB matrix?

Here are the references I'm currently working with:
http://picamera.readthedocs.org/en/release-1.9/recipes2.html#unencoded-image-capture-rgb-format
http://picamera.readthedocs.org/en/release-1.9/array.html#picamera.array.PiRGBArray

waveform80, 2015-01-26:
Hi Lee,

Sorry - I did see your e-mail but I'm a bit swamped with work at the moment, and haven't got much time for responding to picamera issues. Still, your one does look like a relatively simple one so let's see if we can deal with it quickly. Here's the code you originally sent in your e-mail:

```python
import picamera
import picamera.array

with picamera.PiCamera() as camera:
    with picamera.array.PiRGBArray(camera) as output:
        camera.capture(output, 'rgb')
        print('Captured %dx%d image' % (
            output.array.shape[1], output.array.shape[0]))
```

As you've noted, that outputs something like `Captured 1366x768 image`. What you may not have noticed is that PiRGBArray *does* provide a 3D matrix, but the above code is ignoring the third dimension (`output.array.shape[2]` would return 3). Specifically, the array is organized as `(rows, columns, color)`, so array[0, 0, 0] is the red component of the first pixel, array[0, 0, 1] is the green component of the first pixel, and so on.

Hence, we can extend the recipe above to do the following:

```python
import picamera
import picamera.array

with picamera.PiCamera() as camera:
    with picamera.array.PiRGBArray(camera) as output:
        camera.capture(output, 'rgb')
        print('Average of red channel: %.2f' % output.array[:, :, 0].mean())
        print('Average of green channel: %.2f' % output.array[:, :, 1].mean())
        print('Average of blue channel: %.2f' % output.array[:, :, 2].mean())
```

If you want to output the entire matrix you could simply do:

```python
import picamera
import picamera.array

with picamera.PiCamera() as camera:
    with picamera.array.PiRGBArray(camera) as output:
        camera.capture(output, 'rgb')
        print(output.array)
```

In which case you should see something like the following (which fairly clearly shows the three dimensional structure of `output.array`):

```
[[[200 227 157]
  [205 232 162]
  [205 231 164]
  ..., 
  [ 88 104  59]
  [ 91 104  63]
  [ 93 105  64]]

 [[205 232 162]
  [205 232 162]
  [205 231 164]
  ..., 
  [ 92 107  63]
  [ 93 105  64]
  [ 93 105  64]]

 [[205 231 172]
  [202 228 170]
  [201 227 164]
  ..., 
  [ 94 107  61]
  [ 91 105  60]
  [ 94 107  63]]

 ..., 
 [[  0   0   0]
  [  0   0   0]
  [  0   0   0]
  ..., 
  [  6  11   0]
  [  4  13   0]
  [  0  10   0]]

 [[  0   0   0]
  [  0   0   0]
  [  0   0   0]
  ..., 
  [  6  11   0]
  [  6  11   0]
  [  5  10   0]]

 [[  0   0   0]
  [  0   0   0]
  [  0   0   0]
  ..., 
  [  6  11   0]
  [  7  12   0]
  [  6  11   0]]]
```

LeeKnudtson13, 2015-01-28:
Thank you for your response.  Another question, hopefully just as easy to answer.  Say I'm taking a picture at 100x100 resolution.. how can I view the array for all RGB values for each pixel?  I expect to produce a file containing ~10,000 1 by 3 matrices in the form [R,G,B].

waveform80, 2015-01-28:
Hi Lee,

I'd recommend having a read of the [numpy docs](http://www.numpy.org/); there're numerous methods provided by numpy for producing output from an array including [numpy.save](http://docs.scipy.org/doc/numpy/reference/generated/numpy.save.html#numpy.save), [numpy.savetxt](http://docs.scipy.org/doc/numpy/reference/generated/numpy.savetxt.html#numpy.savetxt) and the array's [tofile](http://docs.scipy.org/doc/numpy/reference/generated/numpy.ndarray.tofile.html#numpy.ndarray.tofile) method.

You can see from my response above that accessing the array is fairly trivial; simply access `output.array` and slice/access it however you like. By far the most efficient methods of writing the array to disk will be found within numpy itself (some of which I've linked to above) because they'll be implemented in C, like most of numpy. However, if those routines can't (directly) be used to produce the precise output format you require (your specification is a little light on detail), it wouldn't be difficult to whip up something yourself. For example, to write out something roughly in the form you've described (just demonstrating with an empty array for simplicity):

```python
from __future__ import unicode_literals

import io
import numpy as np

a = np.zeros((100, 100, 3))
with io.open('output.txt', 'w') as f:
    for row in a:
        for pixel in row:
            r, g, b = pixel
            f.write('[%d, %d, %d]' % (r, g, b))
```

closed 2015-03-22 by waveform80