waveform80/picamera #189: can't do capture with resize after one without resize

achrn, 2015-01-31:
I'm having problems with resize in python3 picamera when capturing stills and interleaving some with resize and some without.

Specifically, it seems that I can't do a camera.capture with resize after I've done one without resize, unless I power down the camera.

That is, this works fine:
```
#!/usr/bin/python3

import time
import picamera
import datetime

with picamera.PiCamera() as camera:

    time.sleep(2)
    print ("1")
    camera.resolution = (2592, 1944)
    camera.capture('first.jpg', resize=(1296, 972))

    time.sleep(2)
    print ("2")
    camera.resolution = (2592, 1944)
    camera.capture('second.jpg', resize=(648, 486))
```
but this, where I first of all do a capture without a resize parameter, fails (even though it's actually generating the same size picture, but this time via camera.resolution).
```
#!/usr/bin/python3

import time
import picamera
import datetime

with picamera.PiCamera() as camera:

    time.sleep(2)
    print ("1")
    camera.resolution = (1296, 972)
    camera.capture('first.jpg')

    time.sleep(2)
    print ("2")
    camera.resolution = (2592, 1944)
    camera.capture('second.jpg', resize=(648, 486))
```

I get:
```
1
2
mmal: mmal_vc_port_enable: failed to enable port vc.ril.resize:in:0(I420): ENOMEM
mmal: mmal_port_enable: failed to enable connected port (vc.ril.resize:in:0(I420))0x2103d80 (ENOMEM)
mmal: mmal_connection_enable: output port couldn't be enabled
```
and then it just hangs (indefinitely, or at least until my patience runs out and I press ctrl-c).

There are clearly workarounds - I can do all my sizing by specifying camera.resolution, or put a resize in every capture (including a resize to the same size - that works), or just capture everything at full resolution and resize by using something outside picamera, but the latter at least seems sub-optimal.

It's Raspbian and firmware is up-to-date:
```
x@pi1 ~ $ uname -a
Linux pi1 3.18.4+ #743 PREEMPT Wed Jan 28 17:14:53 GMT 2015 armv6l GNU/Linux
x@pi1 ~ $ /opt/vc/bin/vcgencmd version
Jan 28 2015 18:21:01 
Copyright (c) 2012 Broadcom
version 57b27a49f21df81f6bc0a0c801bd62e94c8e54bd (clean) (release)
```

waveform80, 2015-03-22:
Sorry for the long delay in getting to this. I can definitely confirm the issue, but I'm not sure what's going on under the covers at the moment. The videocore log shows the following when the issue manifests:

```
995334.047: alloc_compact_internal(space=7589440, in_low_region=0, mode=7) failed
995334.074: mmal: mmal_ril_port_payload_alloc: could not allocate 7589376 bytes
995334.103: mmal: mmal_pool_initialise_buffer_headers: failed to allocate payload 6/10
995334.163: mmal: mmal_port_connection_enable: failed to create pool for connection
995334.190: mmalsrv: mmal_do_port_enable: could not enable port (1:ENOMEM)
```

So, it's failing when attempting to allocate 7589376 bytes. That's the amount of memory required to hold a 2592x1952 YUV image (1952 because 1944 is rounded up to a multiple of 16) so it's trying to allocate a buffer to hold a full resolution image in YUV format. Why it fails to do this after a capture with no resize I'm not sure, but something's evidently leaking or left over.

GuardRex, 2015-03-22:
Just wanted to note from my (probably) related issue #150 that Dave's suggestion to me to add a `resize=(x,y)` to every `camera.capture()` call in the app, whether it was needed or not, solved my issue of receiving the exact same exception.

waveform80, 2015-03-22:
A bit more information on this:

1. The issue occurs when the connection between the camera port and resizer is created and enabled
2. This feels as though it's related to #73, so I patched `PiEncoder._create_connections` to look at the various buffer sizes allocated:

```patch
diff --git a/picamera/encoders.py b/picamera/encoders.py
index e9c52b9..f915393 100644
--- a/picamera/encoders.py
+++ b/picamera/encoders.py
@@ -647,6 +647,12 @@ class PiEncoder(object):
         :attr:`encoder_connection` and :attr:`resizer_connection` attributes as
         required.
         """
+        print('Input port: %dx%d' % (self.input_port[0].buffer_num, self.input_port[0].buffer_size))
+        if self.resizer:
+            print('Resizer input: %dx%d' % (self.resizer[0].input[0][0].buffer_num, self.resizer[0].input[0][0].buffer_size))
+            print('Resizer output: %dx%d' % (self.resizer[0].output[0][0].buffer_num, self.resizer[0].output[0][0].buffer_size))
+        print('Encoder input: %dx%d' % (self.encoder[0].input[0][0].buffer_num, self.encoder[0].input[0][0].buffer_size))
+        print('Encoder output: %dx%d' % (self.encoder[0].output[0][0].buffer_num, self.encoder[0].output[0][0].buffer_size))
         assert not self.encoder_connection
         if self.resizer:
             self.resizer_connection = self.parent._connect_ports(
```

Here's an updated test script which also triggers the issue:

```python
import time
import picamera
import datetime

with picamera.PiCamera() as camera:
    camera.resolution = (2592, 1944)
    camera.framerate = 15
    time.sleep(2)
    print ("1")
    camera.capture('1.jpg', resize=(648, 486))
    print ("2")
    camera.capture('2.jpg')
    print ("3")
    camera.capture('3.jpg', resize=(648, 486))
```

It appears the issue occurs under the following circumstances (which is why I think this is related to #73):

1. The camera must be at full resolution
2. The still port must have been switched to OPAQUE encoding at some point (the second capture in the script above causes this as it's a still port capture without a resizer)
3. Once the still port is switched back to I420 after being in OPAQUE, the ENOMEM error occurs
4. However, interestingly, the buffer sizes output by the test script with the patch in place indicate that (as far as picamera is concerned) the buffer sizes look perfectly sane:

```
(picameraenv)dave@kermit ~/foo $ python 189.py 
1
Input port: 1x7589376
Resizer input: 1x7589376
Resizer output: 1x499968
Encoder input: 1x499968
Encoder output: 1x321408
2
Input port: 3x144
Encoder input: 3x15360
Encoder output: 1x5059584
3
Input port: 1x7589376
Resizer input: 1x7589376
Resizer output: 1x499968
Encoder input: 1x499968
Encoder output: 1x321408
mmal: mmal_vc_port_enable: failed to enable port vc.ril.resize:in:0(I420): ENOMEM
mmal: mmal_port_enable: failed to enable connected port (vc.ril.resize:in:0(I420))0x1bdf510 (ENOMEM)
mmal: mmal_connection_enable: output port couldn't be enabled
```

I'm now swaying toward the opinion that this might be an upstream issue. It looks like something is allocated when we do a capture with the still port in OPAQUE encoding, and when we go back to using the port with I420 encoding, we run out of memory (but the buffer sizes are exactly the same so it must be something else ... something GPU side that we can't see?).

waveform80, 2015-03-22:
It's also worth noting that closing down the camera and re-initializing it between each capture also fixes the issue so whatever it is, it's something associated with the camera component. I don't think this is going to be fixable for 1.10. For now, I'll mark this as upstream and try and put together a C-based test case.

6by9, 2015-04-13:
```mmal_pool_initialise_buffer_headers: failed to allocate payload 6/10```
10 buffers on the link at 7.5MB each? Seems excessive. Normally buffer_num will be 3 for a video port, or 1 for stills. Opaque will update buffer_count_recommended to something higher as those are only references to other buffers and are therefore tiny. 
Is it that buffer_num_recommended isn't being reset after switching back from opaque mode, and you're just taking it at face value?

waveform80, 2015-04-13:
Hi Dave, thanks for chipping in! I don't *think* it's down to `buffer_num_recommended`... Firstly, PiCamera now always relies on `buffer_num_min` instead of recommended (because of raspberrypi/userland#167). Secondly, the patch in the comment above has the script report the buffer num and size of each component right before everything gets connected up and the capture starts, and nothing seems to be requesting 10 buffers.

Just to explain what's going on in the test script a little better:

* The first capture involves a request for a resize. The camera is initialized by PiCamera with the still port in I420 encoding, so it leaves it as that, creates a resizer and an encoder, links it all up and produces the capture (destroying the encoder and resizer afterward)
* The second capture doesn't involve a request for a resize. For this, PiCamera reconfigures the still port in opaque mode, creates the encoder, and produces the capture (again, destroying the encoder afterward)
* The third capture is like the first, so PiCamera once again reconfigures the still port to I420 encoding, creates the resizer and encoder, goes to connect everything up and then something blows up by requesting silly amounts of memory.

The terminology in the output is a bit off though. "Input port" means the camera output port that the encoder is attached to, so in the first and third cases the camera's output port has 1 buffer of 7589376 bytes allocated (unsurprising given its in I420 encoding) and in the second case has 3 buffers of 144 bytes allocated (again, unsurprising given its in opaque mode).

All the other buffer sizes look reasonably sane to me too (and those were all the ones I could think to check: the camera's output port, the resizer's input and output ports, and the encoder's input and output ports). So I'm not sure where this idea to allocate 10 huge buffers is coming from ... although it's just occurred to me I'm not looking at the other ports of the camera (they shouldn't be affected by anything in the test script ... but it might be worth a look).

6by9, 2015-04-14:
Looks like I need to address that bug in setting buffer_num_recommended back when switching out of opaque mode.

https://github.com/raspberrypi/userland/blob/master/interface/mmal/util/mmal_connection.c#L296 in mmal_connection_enable
```
   if (!(connection->flags & MMAL_CONNECTION_FLAG_KEEP_BUFFER_REQUIREMENTS))
   {
      if (out->buffer_num_recommended)
         out->buffer_num = out->buffer_num_recommended;
      if (out->buffer_size_recommended)
         out->buffer_size = out->buffer_size_recommended;
      if (in->buffer_num_recommended)
         in->buffer_num = in->buffer_num_recommended;
      if (in->buffer_size_recommended)
         in->buffer_size = in->buffer_size_recommended;
   }
```
So the connection will take buffer_num_recommended, unless you set the magic flag MMAL_CONNECTION_FLAG_KEEP_BUFFER_REQUIREMENTS (it's a new one on me too!). Really ought to fix the underlying issue, but that should run as a workaround.

waveform80, 2016-05-07:
This is working happily in the 1.11 pre-release (might be working in 1.10 too as I don't think anything in 1.11 specifically fixed it, but too tired to test now :)

closed 2016-05-07 by waveform80