waveform80/picamera #190: Doesn't work on a pi 2

paulhothersall, 2015-02-02:
GPIO calling issue?

waveform80, 2015-02-02:
I've just ordered a Pi 2 so I won't be able to test this for a while but that looks like an upstream issue in the RPi.GPIO library. I know that library tests the version of the Pi it's running on as there's differences between (for example) the Model B Rev 1, Rev2, and B+ GPIOs; at a guess it hasn't been updated for the Pi 2 yet. I suppose I could make the import optional and disable the use of the `led` property if it fails to import, but I'd be surprised if the library doesn't get updated pretty quickly now the Pi 2 is out.

I'll mark this as an upstream bug for the time being and update further when I get a Pi 2 and actually get a chance to test it for myself.

paulhothersall, 2015-02-03:
I can give you VPN / SSH access to one right now if you want.....

I suspect if the GPIO calling is like 

import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)

then the python library maybe just in need of an update and the problem goes away?

bennuttall, 2015-02-03:
Unfortunately RPi.GPIO contains something which depends on the memory being where it is on Pi 1. The author, Ben Croston, is currently working on a fix.

robocoder-db, 2015-02-08:
The problem is probably with lines 82-92 of `picamera/picamera/camera.py`, since the Pi 2 is not one of the possible options here.

waveform80, 2015-02-10:
There's been some movement upstream on this - for adventurous sorts, a development version of [RPi.GPIO can now be installed](http://sourceforge.net/p/raspberry-gpio-python/wiki/install/) which should enable picamera to work normally on a Pi2. I'll wait until this is packaged up and released then update picamera's requirements to depend on the released version to ensure the next version guarantees Pi2 compatibility.

kenkam, 2015-02-15:
It looks like https://pypi.python.org/pypi/RPi.GPIO/0.5.10 fixes this issue, and it looks like Raspbian has updated this in the repository.

    $ sudo apt-get update && sudo apt-get upgrade

paulhothersall, 2015-02-15:
confirmed.

looks like its fixed now!

closed 2015-02-15 by paulhothersall

waveform80, 2015-02-15:
Excellent news - thanks to everyone for the comments and updates on this ticket! I've just (briefly) tested this on my Pi 2 and it seems happy (including controlling the LED, so I guess RPi.GPIO has stuck with treating the Model B+ and Pi 2 as equivalent for RPI_REVISION); I'll get on with running the full test suite in a bit.