waveform80/picamera #191: Picamera web interface (mjpeg streaming, etc)

spikedrba, 2015-02-03:
Dave, all

thank you so much for all the amazing work.

I've been working on a security system RPi based and I'm using this code based on picamera to get motion detection and recording done (works like a charm!):

https://gist.github.com/spikedrba/aecb82d8b51e991dbd01

I'm now trying to figure out the best way to expose the stream and recorded videos over the web very much like it's being done in this project:

https://github.com/silvanmelchior/RPi_Cam_Web_Interface

The second part is easy, but exposing the stream is being an issue. H264 is very efficient, but there doesn't seem to be a good way to get that going for streaming in a browser, which I guess is why most IP cameras use MJPEG. I know picamera has support for that in start_recording, however dumping that to a network stream as described in the  _Recording to a network stream_ basic recipe is only giving me one image and then an error (using an html img tag to load it).

I could alternatively use capture_sequence somehow and output stills to the same file name, which is what the above project does (in a ramdisk) and then the web interface reads it out again into a img src tag. The problem with this and the existing code I have is that it would be blocking the script (compared to start_recording). I could put that in a thread, but I'm not sure it's a good solution.

The last idea I had was to use a custom output that would take a h264 stream and output every X frames (depending on fps) to disk as jpg (basically becoming a non blocking capture_sequence).

Am I looking at this completely wrong? Happy to put in the time, but would appreciate some pointers/direction.

thanks,

Spike

paulhothersall, 2015-02-04:
I am having a somewhat similar issue. In my case the stream circular buffer does't work as I loose frames on resuming back to the stream (no lock especially does this) . I have this need as i want to potentially read the buffer out several times with overlapping video clips to create.

I made my own sort of sequence to 1 seconds clips (a rotating 1 minute buffer) in a RAM drive using split stream. This collection of files can be shared quite easily as a network drive, and read or other manipulated away from picamera as to whatever you need. At the expense of a 1 seconds latency! and you are abel to do so out of sequence to capturing frames in the main thread.

I suppose I could learn hoot setup multiple circularIO buffers, but that ends up being way more complex.

spikedrba, 2015-02-04:
@paulhothersall , thanks for chiming in. What do you mean circular buffers in your case lose frames? even more interesting, how can you tell? I was looking for ways to analyze the stream (dump all the frames, their info etc, but couldn't find a lib/example to do that.

I'd love to take a look at your code if you can share it, I'm defintiely ok with 1s latency for the web stream.

thanks, Spike

paulhothersall, 2015-02-05:
http://petrkout.com/electronics/low-latency-0-4-s-video-streaming-from-raspberry-pi-mjpeg-streamer-opencv/

its not using picamera, but will give you you the idea about what is going on for the web interface part.

the buffers have an issue in the sense if you lock it it takes to long to read to even a ram drive. No locking and you can get the stream mixed up.....which is bad.

spikedrba, 2015-02-05:
@paulhothersall I used to use mjpeg-streamer and in the end switched to raspimjpeg and RPi_Cam_Web_Interface:

- https://github.com/silvanmelchior/userland
- https://github.com/silvanmelchior/RPi_Cam_Web_Interface

I get a lot better performances out of those and a nice web interface. The problem is raspimjpeg has no buffer so often the video is missing the first part of the motion.

Spike

waveform80, 2015-03-22:
Writing a chapter on web streaming is something I need to spend some time on, but I'll just spend a moment responding to some of the points in the discussion above:

H.264 is easily the best format in terms of quality and size for streaming. However, most HTML5 browser clients expect a nicely encapsulated MP4 video stream for playing. Remember that the Pi's camera module is, in essence, a stripped out mobile phone camera. In a mobile phone firmware you don't want the camera to produce a full MP4 video - you want it to produce the raw H.264 video stream so that you can mux it with, say an AAC audio stream from the microphone to produce a nice MP4 video. So that's exactly what the Pi's camera gives you: a raw H.264 video stream. It's up to you to encapsulate it nicely for browser playback.

At the moment the best/easiest way of wrapping the H.264 output into an MP4 is with the MP4Box utility. However, this only works on files: not streams. At the moment I'm not aware of a good (read as: simple to implement) method of encapsulating a stream. If someone's aware of one, let me know and I'll add it to the docs.

On MJPEG, I'm not surprised that pointing an `<img>` tag at it only displays the first frame. It's probably interpreting it as a JPEG file with a load of garbage on the end. The `<img>` tag is generally used for static images - for example, pointing one at an MP4 file doesn't work either. You should probably try the HTML5 `<video>` tag instead or write some client side JavaScript to strip out the individual frames and update an image or canvas element with them.

On streaming latency (not mentioned above, but it comes up so frequently in relation to this topic I may as well stick all the info in one place): a lot people seem to be under the mis-apprehension that H.264 encoding introduces lots of latency. This isn't the case. H.264 *playback* typically introduces lots of latency because almost every video player is designed to buffer lots of data (several seconds worth) to guard against lost packets, network issues, etc. Therefore, getting low latency playback generally involves persuading the *client* to forgo its buffering (this is easier said than done, and again I've no idea how to do in the case of HTML5 browsers - any tips welcome!)

In the meantime people may want to have a look at the [pistreaming demo](https://github.com/waveform80/pistreaming) I put together which uses ffmpeg to transcode to MPEG1, and uses an open source JavaScript based MPEG1 decoder (yes, horrid, but it works and it's near-universally supported) to provide a live stream from the camera. Feel free to fork and extend it however you like - I'm happy to accept pull requests with enhancements!

spikedrba, 2015-03-26:
Dave,

thanks for chiming in, very much appreciated. And thanks for the pistreaming demo, it gave me a great starting point to integrate my motion detection script so that I can have a low res streaming while a full res video is recorded and a third stream is used for motion detection with another custom output (using the 1-3 splitter ports available).

Regarding the MJPEG comment, I guess I misunderstood what it was, in my mind I thought of it as "a series of jpeg compressed images" and so I could just dump it in there, but I see that's not right.

So let me ask a different question: would it work to write a custom output that parsed the h264 stream and wrote out jpeg encoded images from all i-frames? this would basically do the same thing as a capture sequence except in a non blocking fashion just like start_recording does and writing to the same file on a ramdisk that I can then serve. This would also allow me to easily do all kinds of reverse proxying and whatnot on that file to serve it to multiple clients from a different machine and leave the Pi alone.

Last but not least, in the pistreaming demo README you mention: "Better still there's plenty of applications which will happily decode and play the H.264 encoded video streams which the Pi's camera produces". Maybe my google-fu is weak, but I couldn't find much that is programmable, like a python library that reads and plays the stream. Closest I could get was gaming related libraries like pyglet. Otherwise maybe vlc which would be installed everywhere and driven with vlc.py bindings. Mplayer has something similar but looks unmaintained. Do you have any recommendation if I wanted to write a desktop client in python to play the videos of 1 or more PIs?

thanks,

Spike

waveform80, 2015-03-26:
Hi Spike,

> Regarding the MJPEG comment, I guess I misunderstood what it was, in my mind I thought of it as "a series of jpeg compressed images" and so I could just dump it in there, but I see that's not right.

Well, you're absolutely correct that it is "a series of jpeg compressed images" - that's a pretty much precise description of MJPEG. It's just a whole bunch of JPEGs concatenated together. The reason that won't display as a video in an `<img>` tag is because that tag is for displaying a *static image* so the browser reads the first JPEG, says "job done" and ignores the rest.

> So let me ask a different question: would it work to write a custom output that parsed the h264 stream and wrote out jpeg encoded images from all i-frames? this would basically do the same thing as [...]

I certainly wouldn't bother with decoding H.264 and then re-encoding it as JPEG images. There's no point when you can easily grab each JPEG image out of the MJPEG stream. Here's [a little gist](https://gist.github.com/waveform80/263b9c8bdcb1e9b79749) I whipped for someone on IRC the other day which uses a [custom output](http://picamera.readthedocs.org/en/release-1.9/recipes2.html#custom-outputs) to extract individual JPEGs from an MJPEG stream (this is easy because JPEG has an extremely well designed ["magic number"](https://en.wikipedia.org/wiki/Magic_number_%28programming%29) (FF D8) which due to the way JPEG is encoded cannot appear within the rest of the JPEG output.

> Last but not least, in the pistreaming demo README you mention: "Better still there's plenty of applications which will happily decode and play the H.264 encoded video streams which the Pi's camera [...]

That's referring to applications like VLC, mplayer, XBMC (aka Kodi), and ffmpeg. However, you're correct there are precious few libraries with decent Python bindings for extracting individual images. OpenCV has a [VideoCapture class](http://docs.opencv.org/modules/highgui/doc/reading_and_writing_images_and_video.html#videocapture-videocapture) which can do this (and which does work with network streams if you pass it an appropriately formatted URL), but all it's actually doing under the covers is firing up an instance of ffmpeg and reading its output (in other words, nothing you couldn't do with `subprocess` in Python). Also, ffmpeg isn't (yet) optimized to use the GPU H.264 encoder/decoder on the Pi so it's slow as anything (in other words, there's no point going that route unless you're doing the decoding on a more hefty machine).

At the moment, your best option for high performance H.264 decoding on the Pi is to dig into omxplayer's source code to learn how to use the GPU's H.264 decoder via OpenMAX (it's probably possible from MMAL as well but I haven't had time to look into that). I can't say that's a terribly easy option though!

previ, 2015-04-07:
@spikedrba, @waveform80: I found a way to stream MJPEG in a browser, you can use a <div> tag and assign to it the following css class:

.ui-content-stream {
background:url('http://<your_mjpeg_stream_url>') no-repeat;
background-size: auto 100% !important;
background-position: center;
}

It works fine in Chrome on Win, OSX and Android; on iOS only Safari is actually working.
Latency is quite low on model B and B+ (about 0.5s@640x480px), even less on a B 2.

waveform80, 2015-04-07:
Interesting stuff (I'm not at all sure why it works though - I would've though CSS `background-image` would interpret things the same as `<img>`! Anyway I'll have a play with that when I get a sec, and see whether it's worth adding something to the doc recipes (although I'm still rather horrified we're resorting to 20 year old formats for streaming!).

spikedrba, 2015-04-08:
@previ pretty excited that this works for you (equally puzzled why it does compared to an <img> tag), but it won't work for me on linux (neither FF nor chrome). In both cases I just get a blank page.

Here's my test code in case you can spot something dumb I'm doing:

https://gist.github.com/spikedrba/38a90e76cfce43c5e358

thanks,

Spike

previ, 2015-04-08:
@spikedrba looking at your code, it seems the HTTP header part is missing.
After the socket is connected, write the following:

            self.send_response(200)
            #self.send_header("Connection", "close")
            self.send_header("Max-Age", "0")
            self.send_header("Expires", "0")
            self.send_header("Cache-Control", "no-cache, private")
            self.send_header("Pragma", "no-cache")
            self.send_header("Content-Type", "multipart/x-mixed-replace; boundary=--BOUNDARYSTRING")
            self.end_headers()

and for each image (frame):

                        self.wfile.write("--BOUNDARYSTRING\r\n")
                        self.send_header("Content-type", "image/jpeg")
                        self.send_header("Content-Length", str(len(_jpegstreamers[port].jpgdata)))
                        self.end_headers()
                        #_jpegstreamers[port]._lock.acquire()
                        self.wfile.write(_jpegstreamers[port].jpgdata)
                        #_jpegstreamers[port]._lock.release()
                        self.wfile.write("\r\n")

(full source file here: https://github.com/CoderBotOrg/coderbot/blob/develop/viz/streamer.py )

@waveform80 I totally agree, MJPEG is definitely not the smartest streaming technology around; a modern and more efficient codec like H264 would lead to better performance, especially in term of bandwidth used.
The problem is, at least for me, that it seems to be quite difficult to implement.

spikedrba, 2015-04-08:
@previ 
bear with me, but I'm confused how this is streaming the actaul MJPEG stream picamera.PiCamera.start_recording gives you back.

The way I read that code is that you expect individual jpeg images, not the stream. If that's the case, or even if it wasn't, what's the advantage over doing something along the lines of this gist @waveform80 put together:

https://gist.github.com/waveform80/263b9c8bdcb1e9b79749

and simply writing to the same file and serving that? (this is what @silvanmelchior's rpi_web_interface does). So basically you pick apart the MJPEG stream and rewrite the img file (doing this in a ramdisk for less lag) and then just serve that file (with lighty or something, or even python, but you're basically just returning a static file). And on the client side you just have a piece of js with a settimeout that reloads the source of the img.

is that a bad idea?

thanks,

Spike

previ, 2015-04-08:
@spikedrba sorry I didn't link the code that does the picamera image grabbing stuff. 
Yes my implementation basically grabs single jpegs (in memory, using BytesIO) and serves them through the HTTPStreamer.
On the client side, no js code is required since the browser is able to process the stream correcly, using the css in the message above (didn't try the <img> tag).
I didn't try to send the MJPEG stream directly to the client, but as far as I understood, this doesn't seem to work because the client is unable to process correctly the MJPEG stream.

waveform80, 2016-09-30:
Many thanks for all the info everyone's provided on this ticket! There'll be an MJPEG streaming demo in the Advanced Recipes section of the next picamera as a result.

commit c92f65e0 (2016-09-30): Fix #191
closed 2016-09-30 by waveform80