waveform80/picamera #192: Rapid execution of split_recording having unpredictable behaviour

paulhothersall, 2015-02-05:
I am having some issues with the split streaming option, in the sense it doesn't always work

see below for code

import picamera,time, io

with picamera.PiCamera() as camera:
    
    camera.resolution = (1280, 720)
    camera.framerate = 30
    
    camera.start_preview()
    camera.start_recording('throwaway.h264', splitter_port=1, format='h264', bitrate=17000000, intra_period=int(30))

    try:
        while True:
            current_second = time.strftime('%S')
            what_file_name = '/mnt/ramdisk/' + current_second + '.h264'
            camera.split_recording(what_file_name, splitter_port=1)

I know this is an ugly hack (generating rotating 1 second clips of the last minute), but works fine, UNTIL you up the bitrate or  or other like multiple streams. then it ends up with a file every 2 seconds (with 60frames) most of the time, apart from when it get just one.

Is there some sort of minimum time in split_recording that I am running up against. For reference the time to write to the file like object is << 1 second, so its not that....

waveform80, 2015-03-22:
I've tried the example script which, as you've mentioned, works happily. I tried raising the bitrate of the encoder to 25Mbps (which is the maximum for the H.264 encoder) but it still happily ran with split points every second. There were odd occasions where it skipped a few seconds, but these turned out to be IO related (probably Linux deciding to flush the write cache at that point) and that's not really surprising given the amount of data being written.

You've mentioned "other streams". Do you mean simultaneously recording to other streams? I tried the following script which records two streams simultaneously, but it also managed to split the first one each second with trouble:

```python
import picamera,time, io

with picamera.PiCamera() as camera:

    camera.resolution = (1280, 720)
    camera.framerate = 30

    camera.start_preview()
    camera.start_recording(
            'throwaway.h264',
            splitter_port=1, format='h264', intra_period=30)
    camera.start_recording(
            'stream2.h264',
            splitter_port=2, format='h264', intra_period=30)

    while True:
        current_second = time.strftime('%S')
        what_file_name = current_second + '.h264'
        camera.split_recording(what_file_name, splitter_port=1)
```

In short, I can't seem to reproduce the issue.

spikedrba, 2015-04-15:
not sure if this is related, but I just ran into this error which sounds the same as @paulhothersall was talking about:

```
Traceback (most recent call last):
  File "mypmd.py", line 309, in <module>
    camera.split_recording(motion_filename + '-after.h264', splitter_port=1)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1301, in split_recording
    encoder.split(output, options.get('motion_output'))
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 1266, in split
    'Timed out waiting for a split point')
picamera.exc.PiCameraRuntimeError: Timed out waiting for a split point
```
this is the code I'm using at the line where it's failing:
https://gist.github.com/spikedrba/aecb82d8b51e991dbd01#file-pmd-py-L308

the one that was actually running at fps set to 45 from the usual 30 and ran for about 6 hours recording  70 videos on motion detection before failing.

LeeKnudtson13, 2015-04-15:
?Please remove me from this mailing list.

Thanks,

Lee

https://gist.github.com/spikedrba/aecb82d8b51e991dbd01#file-pmd-py-L308

waveform80, 2015-04-15:
@LeeKnudtson13 - I can't remove you from notification list. You can easily do this yourself simply by visiting the picamera project on GitHub and clicking on "Unwatch" at the top (you added yourself to the watchers list which is why you're seeing notifications for everything going on).

waveform80, 2016-05-07:
Minimum timeout for the split-point is raised to 10 seconds in 1.11; so far I haven't seen a recurrence of this issue in the test suite (used to be an occasional thing on slower platforms like the 1B).

closed 2016-05-07 by waveform80

mohsin, 2022-11-27:
I’m facing the same issue on [my project](https://github.com/mohsin/cat-detection/issues/1). Using a raspberry pi zero w. Is my settings bad?