waveform80/picamera #193: Add intra-refresh type property

waveform80, 2015-02-09:
There's a new option pending in raspivid to control intra-refresh type; not sure what that is yet, but we should have an equivalent property in picamera.

JamesH65, 2015-02-10:
There are a number of different ways of formatting the I-frame, this gives the option to specify which one you want. Not exactly sure what some do, but cyclic rows can be used to partially send bits of an I-frame over time - which can smooth out bandwidth requirements, but with a slight compression penalty.

waveform80, 2015-02-10:
Ah - sounds handy for streaming which is something I desperately need to write a FAQ about (the number of questions I get about streaming seems endless!)

commit 659df215 (2015-03-11): Fix #193
closed 2015-03-11 by waveform80