waveform80/picamera #195: Intermittent PiCameraValueError raised when using picamera.array.PiRGBArray and KeyboardInterrupt is raised

kenkam, 2015-02-15:
I have intermittently encountered a ``PiCameraValueError` when I Ctrl-C out of my application that is capturing images in a ``while True`` loop. I'm using ``PiRGBArray``. It looks like this happens when the Ctrl-C occurs before all the bytes have been received by the encoder and ``encoder.stop()`` is called, meaning when ``flush`` is called on the stream `bytes_to_rgb`` has less data than it expects.

Not entirely sure whether this needs to be fixed, and if so, where the best place to fix it is. Perhaps we could add an optional parameter in ``PiEncoder._close_output(suppress_value_errors=False)`` that catches and suppresses  ``PiCameraValueError`` errors raised if ``suppress_value_errors == True``? 

Thanks,
Ken

Stacktrace:

    File "./motion.py", line 136, in <module>
      motion_detector.run()
    File "./motion.py", line 91, in run
      next_img = self.camera.capture_opencv_image()
    File "./motion.py", line 33, in capture_opencv_image
      self.camera.capture(stream, use_video_port=True, format='bgr')
    File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1466, in capture
      encoder.close()
    File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 758, in close
      self.stop()
    File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 746, in stop
      self._close_output()
    File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 660, in _close_output
      output.flush()
    File "/usr/lib/python2.7/dist-packages/picamera/array.py", line 275, in flush
      self.array = bytes_to_rgb(self.getvalue(), self.size or self.camera.resolution)
    File "/usr/lib/python2.7/dist-packages/picamera/array.py", line 173, in bytes_to_rgb
      'Incorrect buffer length for resolution %dx%d' % (width, height))
    picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 320x240

waveform80, 2015-03-22:
Hi Ken,

Sorry for the long delay in getting to this. The problem is that when you hit Ctrl+C, sooner or later the main thread stops the recording. When stopping the recording, the encoder will attempt to flush the output object. The stream-array classes use the `flush` call to determine when to try and construct an array from the output. In the event of a partially written buffer, the stream-array classes raise the `PiCameraValueError`.

In other words, the PiCameraValueError is a secondary exception raised in response to the KeyboardInterrupt exception. This is quite normal and not a bug (it's more a consequence of how generic the output mechanism is in picamera), but it is difficult to detect and respond to in Python 2 (the usual method there would probably be to catch the `PiCameraValueError` as well in whatever loop your main script runs). Under Python 3, [exception chaining](http://stackoverflow.com/questions/16414744/python-exception-chaining) is implemented so you could potentially look back through the stack of exceptions to find the `KeyboardInterrupt` one.

Basically this isn't something I can easily fix in picamera, and certainly not in a way that is compatible with all python versions that are currently in the wild. Sorry!

manoelnetom, 2015-04-28:
you need to clear down the stream between captures. The PiRGBArray docs cover doing this with truncate function see http://picamera.readthedocs.org/en/release-1.7/array.html#picamera.array.PiRGBArray

kenkam, 2015-07-09:
Ah, thanks for answering. That makes perfect sense. I'll catch the exception then.

closed 2015-07-09 by kenkam