waveform80/picamera #197: Black screen when adding multiple overlays to the screen

glamorous, 2015-02-21:
I've some strange problems with adding different overlays at the "same" time but I'm not receiving any errors or so.

What I'm doing:

    cam.create_overlay('start', LOCATION_OVERLAYS + 'overlay_push.jpg', layer=5)
    cam.create_overlay('3', LOCATION_OVERLAYS + 'overlay_3.jpg', layer=4)
    cam.create_overlay('2', LOCATION_OVERLAYS + 'overlay_2.jpg', layer=3)
    cam.create_overlay('1', LOCATION_OVERLAYS + 'overlay_1.jpg', layer=2)
    cam.create_overlay('bg', LOCATION_OVERLAYS + 'overlay_bg.jpg', resolution=(0, 0, 1920, 1080), layer=0)
    cam.create_overlay('smile', LOCATION_OVERLAYS + 'overlay_smile.jpg', resolution=(0, 0, 1920, 1080), layer=8)
    
    cam.show_overlay('start')
    cam.show_overlay('bg')
    cam.start()

with 
- `cam.create_overlay` is `camera.create_overlay` call with alpha 0
- `cam.show_overlay` is setting the overlay it's `alpha`-value to 255
- `cam.start()` is `camera.start_preview(fullscreen = False, window = (0, 0, 1620, 1080), layer = 1)`

I'm seeing some overlay-images flash and then blank until the script finishes. The red led from the camera is on. When I comment out a couple of overlay-lines (randomly), then I can see the preview and the 2 or 3 overlays that are active.

Anyone knowing what I'm doing wrong?

waveform80, 2015-02-21:
Can you include the code for `create_overlay` and `show_overlay` (and possibly `start` if it's relevant)? Those aren't bits of picamera, so I need to know exactly what they're doing to figure this out. It'd also be useful to know what the resolutions of the various JPEG files are.

glamorous, 2015-02-22:
def create_overlay(self, name, filename, resolution=False, layer=3):
        img = Image.open(filename)
        # Create an image padded to the required size with
        # mode 'RGB'
        pad = Image.new('RGB', (
            ((img.size[0] + 31) // 32) * 32,
            ((img.size[1] + 15) // 16) * 16,
            ))
        # Paste the original image into the padded one
        pad.paste(img, (0, 0))
        window_resolution = self.overlay_sizelocation if (resolution == False) else resolution
        o = self.cam.add_overlay(pad.tostring(), size=img.size, fullscreen=False, window=window_resolution, alpha=0, layer=layer)
        self.overlays[name] = o

with `self.overlay_sizelocation` = `(1626, 0, 288, 656)`

    def show_overlay(self, name):
        self.overlays[name].alpha = 255

    def start(self):
        self.cam.start_preview(fullscreen = False, window = (0, 0, 1620, 1080), layer = 1)

All my JPEG's:
- overlay_1.jpg - 7kb - 288x656
- overlay_2.jpg - 13kb - 288x656
- overlay_3.jpg - 14kb - 288x656
- overlay_push.jpg - 21kb - 288x656
- overlay_bg.jpg - 65kb - 1920x1080
- overlay_smile.jpg - 58kb - 1920x1080

Hope this info is enough :)

glamorous, 2015-02-22:
A little bit more information on the problem:
I just rebooted the device and overlay_1, overlay_2, overlay_3 were commented out (so no creation of those 3 overlays) and everything looked normal. So I thought maybe it's a memory issue and was "fixed" by rebooting the device so I decided to uncomment them and try again. Then nothing appeared, just a flash (like before). Ok, then comment the three out again and test again: and not working, while it was working the first time after reboot...

My code is the initial file is

    cam = camera()
    try:
        # setup booth (create overlays and start preview, code mentioned in first post)
        setup_booth(cam)
        #irrelevant code because when this is commented out, it doesn't matter
    finally:
        print "in finally"
        cam.clean()

With the clean function in cam:

    def clean(self):
        self.cam.close()

So I suppose everytime the script finish or when I finish it, it will close the camera and cleanup the memory?

waveform80, 2015-03-14:
Sorry for the long delay - been a rather busy few weeks (especially with the Pi's birthday bash!). I've had a chance to look into this a bit further, and it looks like there's definitely a limit to the number (and size) of overlays, and that bad things happen when you break that limit. Specifically, it seems that when too much memory has been used up rendering overlays, things go a bit screwy and the Pi tries to give the monitor a bad mode (on my HDMIPi, the screen goes red and reports "out of range", or it starts flashing blue indicating it's trying to set a mode but is failing for some reason). I suspect that much is a firmware issue (i.e. it isn't meant to send bad modes to the monitor, but should report an error instead), so there's not much I can do about that.

However, there is a relatively simple workaround I think you can use in your case. Specifically, overlays don't need to be the same size as their display window; they can be significantly smaller and the overlay system can rescale them on the fly to fill the screen. With two 1080p overlays (as in your example) I could trigger the issue, but when I scaled them both down to 720p (and simply set a larger window position to rescale them back up on the display) it worked happily. Hopefully that should be good enough for your booth?

glamorous, 2015-03-14:
Should I or you report an error to the Pi community for that bug in the firmware? Do you think there would be less issues with a RaspberryPi2 with more memory? I've never had "strange" hdmi outputs like a blue or red screen. I've used the pi on a Sharp 60inch TV for that matter.

I will give it a try with using 720p overlays, hope the quality of the fonts and text in it, or not bad when it's scaled up. But maybe that will not trigger the memory issues. Didn't thought that 50-60kb images would trigger those issues...

Will let you know if that fixes it for me...

waveform80, 2015-03-14:
I'll try a few more tests to see if I can narrow things down a bit more precisely before reporting it upstream. In particular I'm interested in whether increasing the GPU memory split increases the amount/size of overlays that causes the issue. I don't think the Pi2 will make any difference by itself, because the default GPU memory split is 128Mb on both on the Pi1 and the Pi2, but obviously there's more scope for expanding it on the Pi2 with its 1Gb of RAM.

On the subject of strange outputs - that's down to the display itself. Most TV's seem to be terse to the point of silence in the case of signal issues; i.e. they'll simply show a black screen. Monitors (like the HDMIPi) tend to be (slightly) more verbose; evidently in the case of the HDMIPi a red screen is how it reports a problem with the signal, and a blue screen is what it shows upon switching to a new mode.

Finally, the images you're working with may be 50Kb on disk but that's in a compressed format. The GPU needs to work with the uncompressed data and they're *much* larger than that once you load them into an overlay as RGB data. Take your 1920x1080 image: in RGB format that's 1920 * 1080 * 3 bytes = ~6Mb of GPU memory. The total memory requirements for all your overlays is about 14Mb. That may not sound like much, but that's just for the raw data. The GPU will want plenty more on top of that for format conversions and all the other work it does.

glamorous, 2015-03-14:
Thanks for the explanation, will keep you posted!

waveform80, 2015-03-25:
According to upstream this is working as designed. See discussion in raspberrypi/userland#197 for full details. I'll add a warning to the `add_overlay` method docs which mentions the `dispmax_offline` workaround but that's about all I can do this one.

closed 2015-03-25 by waveform80

glamorous, 2015-04-14:
Hi,

Just tested it with the new overlays on 720p, but still having those issues. Is it possible to guide me to the right direction for downscaling and upscaling in my code?

Thanks on beyond!

commit c31a2f5e (2015-03-25): Fix #197