waveform80/picamera #200: Question about shutter speed

Gelu74, 2015-02-24:
Hi Dave,
Thanks for the great picamera woderfull library.
 I was wondering whether it is somewhere documented the stop values for the shutter speed in the camera module. In picamera I can set the shutter_speed in microseconds but it seems to me that these are approximated to the closer real shutter speed value the camera supports.

For example if I run:
In [62]: camera.shutter_speed=410

In [63]: camera.exposure_speed
Out[63]: 401L

In [64]: camera.shutter_speed=401

In [65]: camera.exposure_speed
Out[65]: 401L

both speeds are equal although the microseconds specified are different.
I was particularly interested in knowing the fastest possible shutter speed.

Regards from Spain.
Angel

6by9, 2015-03-03:
The sensor sets the shutter speed as an integer number of lines read out, hence the rounding. The time taken to read out a line varies based on mode, but the 5MPix mode takes 29.5usecs per line.
Minimum shutter speed appears to be 6 lines, or 177usecs.

BTW Best to ask these questions on the forum (http://www.raspberrypi.org/forums/viewforum.php?f=43) as that is where those who know about the firmware lurk. Picamera is waveform80's wrapper around the GPU firmware.

waveform80, 2015-03-03:
Hi Angel,

Just to expand a little on 6by9's answer (which I would consider definitive; as an ex-firmware dev he knows a great deal more about the low level operation of the camera than I), the shutter speed is clamped at the upper end by `framerate` (so it can't be longer than it would take to expose one frame) and at the lower end by line read-out time (which as mentioned above is mode-specific so you'll find it differs when you change `resolution`).

When AGC is running (i.e. when `exposure_mode` is something other than "off") you'll see the `analog_gain` and `digital_gain` increase to compensate for lower speeds. For example, when experimenting with this on my development Pi, I pushed shutter speed down from 13,000µs to 6,000µs. The preview display barely changed brightness because `analog_gain` doubled to compensate. Obviously with extremely fast shutter speeds (low values), the gain limits are eventually hit and the preview/capture becomes darker.

closed 2016-01-19 by Gelu74