waveform80/picamera #203: Update Python Picamera for Annotate_V3

jsh121988, 2015-03-03:
6by9 updated the camera firmware to support ANNOTATE_V3 which gives custom control over the annotation.

I was hoping we could see an update to picamera to support this.I tried to do it, but I'm not very skilled in python yet. The properties would not take/hold a given value, and I'm not sure what the requirements are to get custom objects working. I know what it takes in C#... 

Here is the thread:
http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=97929

Here is the object for ANNOTATE_V3:
#define MMAL_CAMERA_ANNOTATE_MAX_TEXT_LEN_V3 256
typedef struct MMAL_PARAMETER_CAMERA_ANNOTATE_V3_T
{
   MMAL_PARAMETER_HEADER_T hdr;

   MMAL_BOOL_T enable;
   MMAL_BOOL_T show_shutter;
   MMAL_BOOL_T show_analog_gain;
   MMAL_BOOL_T show_lens;
   MMAL_BOOL_T show_caf;
   MMAL_BOOL_T show_motion;
   MMAL_BOOL_T show_frame_num;
   MMAL_BOOL_T enable_text_background;
   MMAL_BOOL_T custom_background_colour;
   uint8_t     custom_background_Y;
   uint8_t     custom_background_U;
   uint8_t     custom_background_V;
   uint8_t     dummy1;
   MMAL_BOOL_T custom_text_colour;
   uint8_t     custom_text_Y;
   uint8_t     custom_text_U;
   uint8_t     custom_text_V;
   uint8_t     text_size;
   char text[MMAL_CAMERA_ANNOTATE_MAX_TEXT_LEN_V3];
} MMAL_PARAMETER_CAMERA_ANNOTATE_V3_T;

waveform80, 2015-03-03:
Ooh, I'd missed that! Okay, that can go on the list for 1.10.

waveform80, 2015-03-16:
Not sure I should mark this closed yet - v3 annotate works fine, but the backward compat stuff for v2 is broken (because the firmware seems completely oblivious to mis-sized structures...)

commit 9d52d00c (2015-03-16): Fix #203
closed 2015-03-16 by waveform80