waveform80/picamera #208: Failed to set preview resolution: Argument is invalid

sabas1080, 2015-04-08:
Hi
I have been working with Picamera bit, but since yesterday I have this error.

Does anyone know who is?

File "cam.py", line 126, in selected                                                                                                      
    else:                  self.callback(self.value)                                                                                        
  File "cam.py", line 311, in viewCallback                                                                                                  
    takePicture()                                                                                                                           
  File "cam.py", line 686, in takePicture                                                                                                   
    camera.resolution = sizeData[sizeMode][0]                                                                                               
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 2159, in _set_resolution                                                 
    framerate=framerate, resolution=(w, h))                                                                                                 
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 2003, in _set_camera_mode                                                
    prefix="Failed to set preview resolution")                                                                                              
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 133, in mmal_check                                                          
    raise PiCameraMMALError(status, prefix)                                                                                                 
picamera.exc.PiCameraMMALError: Failed to set preview resolution: Argument is invalid

LeeKnudtson13, 2015-04-10:
?remove me from emailing list please

waveform80, 2015-04-10:
@LeeKnudtson13 - just click "unwatch" at the top of the page and you'll stop receiving all these annoying e-mails...

waveform80, 2015-04-10:
@sabas1080 - ideally I'd need to see some of your code to figure out what's going on here. The bit that sets `sizeData[sizeMode]` in particular would be handy. I'm guessing that it's something other than a `(width,height)` tuple

sabas1080, 2015-04-11:
if I miss capturing a "while" I work well , if I leave within a "thread" i have errors

waveform80, 2015-04-11:
@sabas1080 the error that's being raised indicates that something invalid is being passed as width and height during `_set_camera_mode` (called by `_set_resolution`) but without a look at the surrounding code I can't guess what might be wrong.

sabas1080, 2015-04-11:
Sorry code , still in draft

https://gist.github.com/sabas1080/88d0a01d94308d6ec3db

line #536 def filename_sec():
line #701 s = threading.Thread(target=filename_sec) #,args=(False,) 

Thanks

waveform80, 2015-04-11:
Hmm, this is a rather complex script (unnecessarily so in several places). I'm not sure what the actual issue is (sizeData seems sane and is never set anywhere except at the start), and the requirements for the script are too complex for me to be able to test it here but my hunch is that the threading is the core issue. Specifically, in `takePicture` (which has several race conditions to do with directory and file creation, but let's ignore that for the time being) you're starting a background thread to take the picture. The background thread switches the resolution of the camera (which involves disabling and re-enabling the camera and therefore has to be done while the camera is completely idle - you can't be recording or capturing while this is going on), but while the background is doing this, you attempt to stop the preview in the main thread.

Basically, threading is a hard topic anyway, and that's especially true when you're dealing with the camera library which (implicitly) starts and stops background threads to deal with capture and recording. As a general rule of thumb, when dealing with picamera (or any library that involves invisible background threads) you should aim to interact with the library from a *single* thread of your own. In other words, create the PiCamera instance, set parameters, capture images, etc. from a single thread.

sabas1080, 2015-04-11:
working solution "splitter_port=1"

camera.capture(filename, splitter_port=1, use_video_port=True, format='jpeg',
		thumbnail=None)

thanks!!

closed 2015-04-11 by sabas1080