waveform80/picamera #209: Problem setting framerate?

brainwagon, 2015-04-08:
First of all, thanks a bunch for writing this awesome library: it allows me to do stuff with the Pi chat I can't even do on much more powerful computers equipped with USB webcams.   

One thing I have been having difficulty with is setting the framerate for video captures.   I've tried setting in both the initializer and after the camera has been created but before start_recording happens:

with picamera.PiCamera(framerate=fractions.Fraction('30/1')) as camera:
    dir = "/var/capture"
    camera.resolution = (1024, 768)
    camera.framerate = fractions.Fraction('30/1')
    print camera.framerate

But when I capture to a file: 
pi@petrichor ~ $ ffmpeg -i /var/capture/235914-A.h264 
ffmpeg version N-66222-g1b5ccae Copyright (c) 2000-2014 the FFmpeg developers
  built on Sep 10 2014 04:14:04 with gcc 4.6 (Debian 4.6.3-14+rpi1)
  configuration: 
  libavutil      54.  7.100 / 54.  7.100
  libavcodec     56.  1.100 / 56.  1.100
  libavformat    56.  4.101 / 56.  4.101
  libavdevice    56.  0.100 / 56.  0.100
  libavfilter     5.  1.100 /  5.  1.100
  libswscale      3.  0.100 /  3.  0.100
  libswresample   1.  1.100 /  1.  1.100
Input #0, h264, from '/var/capture/235914-A.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1024x768, 25 fps, 25 tbr, 1200k tbn, 50 tbc

I get 25fps.  I have no idea what I'm doing wrong.   Any ideas?

Mark

waveform80, 2015-04-11:
Hi Mark,

You're not doing anything wrong. This is a symptom of the fact that the raw H.264 stream output by the Pi's camera module does not contain any information about the capturing frame rate. Anything reading the stream either has to be told the framerate explicitly or guess one; everything that I'm aware of (ffmpeg, VLC, mplayer, MP4Box, etc.) guesses one and most default to 25fps (or something very close). This is why if you record something at <25fps (e.g. 1fps) and play it back, it looks like a timelapse recording and conversely if you record something at 60fps and play it back it looks like slow-mo.

Unfortunately, in the case of ffmpeg it only allows you to specify a framerate on the command line (with the `-r` parameter) when dealing with a "raw" format, and "raw" in this case means "completely unencoded" like a bunch of RGB frames concatenated together. A raw H.264 stream isn't "raw enough" for ffmpeg so you can't specify a framerate for it (very annoying!).

The best I've managed in the past is to use one ffmpeg instance to decode the H.264 stream to raw frames then pipe that to another ffmpeg instance for further processing with an explicit framerate (of course, any kind of serious compression like libx264 runs terribly slowly on the Pi's CPU so this is best done on a beefier machine). Let me know if you need a hint on doing that and I'll dig out the command line - but if you're just wondering why the framerate is reported wrong, there's your reason :)

brainwagon, 2015-04-13:
Thanks for the explanation, I was mystified as to why it was reporting
25fps, but never considered that the actual framerate in the file would be
different.  Makes perfect sense.  I think I'll just be happy with 25fps,
since it's more important for me to do things efficiently than to get an
particular framerate.   I'm using this as the basis for a motion detecting
camera that I'm using to monitor my hummingbird feeder.
http://brainwagon.org/2015/04/10/skeleton-of-a-motion-detecting-video-capture-program-for-the-raspberry-pi-camera/
 Thanks for your help!

closed 2016-05-07 by waveform80