waveform80/picamera #210: possible bug in renderers.py ?

trappn, 2015-04-14:
Hi,
first of all, thanks for this excellent library!
I think I might have found some bug in the function that compares buffer sizes, but have been unable to track it down. You can reproduce it using the simple crosshair / numpy array example (4.15 in the docs). It works fine as-is, and I had no problems changing the resolution to 1024x768. Anything else fails with this error:
```
Traceback (most recent call last):
  File "test.py", line 19, in <module>
    o = camera.add_overlay(np.getbuffer(a), layer=3, alpha=64)
  File "/usr/local/lib/python2.7/dist-packages/picamera/camera.py", line 1102, in add_overlay
    renderer = PiOverlayRenderer(self, source, size, **options)
  File "/usr/local/lib/python2.7/dist-packages/picamera/renderers.py", line 557, in __init__
    self.update(source)
  File "/usr/local/lib/python2.7/dist-packages/picamera/renderers.py", line 578, in update
    sp = bp.from_buffer(source)
ValueError: Buffer size too small (1440000 instead of at least 1459200 bytes)
```
It's probably a rounding problem somewhere (note that 1440000 = 800x600x3).

6by9, 2015-04-14:
(Dave, hope you don't mind me jumping in, I just happened to be reading the issues page :-) )
To quote the docs
> One difficulty of working with overlay renderers is that they expect unencoded RGB input which is padded up to the camera’s block size. The camera’s block size is 32x16 so any image data provided to a renderer must have a width which is a multiple of 32, and a height which is a multiple of 16. 

600 = 16 * 37.5, so not a multiple of 16, therefore it fails.
1459200 = 800 * 608 * 3

trappn, 2015-04-14:
Ah, interesting. Then I'll just implement a check for this as a workaround in my code.
Thanks.

waveform80, 2015-04-15:
@6by9 - not at all, always great to have your input!

@trappn - 6by9's explained the problem, that the data provided to the renderer has to be a multiple of the block size. This doesn't mean the image has to be a multiple of the block size, but that the data representing the image must be. Have a look at the [associated recipe](http://picamera.readthedocs.org/en/release-1.10/recipes1.html#overlaying-images-on-the-preview) for information on how to deal with an image which isn't such a size (just involves padding the image up to the required size, then using the `size` parameter to add_overlay to specify the actual size of the image).

trappn, 2015-04-20:
OK, thanks for the clarification. I'm not sure this solution is really necessary in my case, but I'll give it a shot.

closed 2016-05-07 by waveform80