waveform80/picamera #211: understanding shutter speed and exposure mode

spikedrba, 2015-04-16:
Hi,

I'm probably misunderstanding something here, but I'm puzzled by what I'm seeing by playing around with exposure_mode and framerate.

My basic understanding from the docs and other issues here is that shutter speed is fundamentally capped by the framerate, in other words if you want 30fps the shutter speed can't be slower than 33.3ms. It eludes me however if and what else has an infuence on it.

Look at this output:

```
>>> c.framerate = 30
>>> c.exposure_speed
33164L
```
which makes sense, however
```
>>> c.framerate = 10
>>> c.exposure_speed
36159L
>>> c.framerate = 1
>>> c.exposure_speed
36322L
```
how is it possible that 1/30th of the fps result in only 3ms slower ss? I'm guessing something else is happening, maybe around light calculation? obviously exposing for a long time in bright light would give you an overexposed picture, even tho maybe the AGC would kick in?

And speaking of light and exposure mode, I've read/was told that night mode fundamentally reduces the number of fps (what else does it do? couldn't find any clear documentation about it), but I couldn't find direct evidence of this:

```
>>> c.exposure_mode = 'auto'
>>> c.framerate = 30
>>> c.exposure_speed
33164L
>>> c.exposure_mode = 'night'
>>> c.framerate
Fraction(30, 1)
>>> c.exposure_speed
33164L
>>> c.start_recording('/dev/null', format='h264', splitter_port=1)
>>> c.framerate
Fraction(30, 1)
>>> c.exposure_speed
33164L
```
as you can see after switching to night mode nothing changed. Thinking it might be detecting the light and based its decision on that, I put a cap on the lense, rendering the shot completely dark. Even under that condition nothing changed, fps and exposure_speed remained at the same values.

What's going on here? what am I missing?

thanks,

Spike

spikedrba, 2015-04-16:
As an aside, today I was testing with raspivid using the -set to get a dump on the settings and changing the fps did result in a change in exposure speed, not sure what's going on with picamera:
```
30fps
mmal: Exposure now 33120, analog gain 2048/256, digital gain 511/256
mmal: AWB R=224/256, B=268/256

15fps
mmal: Exposure now 66461, analog gain 2048/256, digital gain 511/256
mmal: AWB R=244/256, B=362/256
```
which is exactly what I'd expect.
thanks

waveform80, 2016-05-07:
Wow - this has taken me a while to get around to. Unfortunately you posted this about two weeks into us having our first child so the timing wasn't ideal! :)

I'm not sure what was going on at the time but I can't reproduce this issue with picamera 1.11 (unreleased at the moment, but there's no real difference regarding exposure speeds between 1.10 and 1.11). Here's a dump of a console log - initially the framerate is 30, but the exposure speed is a mere ~1700µs because the camera's in bright daylight:

```python
>>> from picamera import PiCamera, mmal
>>> camera = PiCamera()
>>> camera.framerate = 30
>>> camera.exposure_speed
1731L
```

Then I drop a box over the camera giving it a very dark scene, measure the exposure speed, adjust the framerate down and measure again:

```python
>>> camera.exposure_speed
33120L
>>> camera.framerate = 15
>>> camera.exposure_speed
39983L
```

Okay, not a huge increase but the box has quite a few holes so it's not completely dark in there. Still, it demonstrates that the firmware is increasing the exposure speed when required.

waveform80, 2017-02-02:
Closing out old tickets, do feel free to re-open if you've still got a query about this!

closed 2017-02-02 by waveform80