waveform80/picamera #213: high cpu usage, am I doing something wrong?

spikedrba, 2015-04-24:
Hi,

I've been using picamera for a while and it's been all kinds of wonderful to get a grib on the workings of the rpi's camera and developing a prototype. I am moveing however toward a more "production" stage and started to pay more attention to performances, especially if I hope to get this going on an A+, which would be great.

The core of what I'm doing surrounds motion detection for which I'm been referencing the following resources:
- https://raw.githubusercontent.com/citrusbyte/pimotion/master/pimotion/pimotion.py
- http://bits.citrusbyte.com/motion-detection-with-raspberry-pi/

from there I've been playing with these two simple scripts:
- https://github.com/citrusbyte/pimotion/blob/master/pimotion/pimotion.py
- https://gist.github.com/spikedrba/aecb82d8b51e991dbd01

in both cases looking at top/vmstat cpu usage on a b+ is close to, respectively, 30% and 40% while not recording and 50% and 70% while recording.

Testing latest raspimjpeg with motion detection (https://github.com/roberttidey/userland/blob/MotionDetection/host_applications/linux/apps/raspicam/RaspiMJPEG.c), base cpu usage is 8% going up to 23% when displaying motion vectors (worth keeping in mind that that seems to be primarily due to the displaying of images containing the vectors requiring conversion from YUV to jpeg, otherwise the analysis itself would not make that much of a diff in cpu usage).

To be clear, I don't mean to make it a python Vs C argument, I'm just looking for input to better understand long term performances of something built around picamera. Am I doing something dumb here that's making those python implementations of motion detection very inefficient?

The baseline isn't too far off, with a simple camera.start_recording taking 10% on the same B+.

Memory profile on the other end seems quite different, with even the simple script taking 3% of the pi's mem up to 10% when doing motion detection while raspimjpeg sits steady below 1%. That said I'm not concerned that much about mem at this point, there's enough and I don't need to keep too much in it except some temp stuff written to a ramdisk.

any input would be much appreciated. Thanks in advance,

Spike

waveform80, 2016-05-07:
As you've noted it's generally unrealistic to expect to get even close to C's performance in Python; that said one can generally get surprisingly close in some situations given that many quick large Python libraries (like numpy) are largely written in C and contain numerous optimizations.

I've had a quick skim through some of your links - in particular your motion detection routines and the similar bits in raspimjpeg. Some things to note:

* Your motion detection involves a lot more heavy lifting than that in raspimjpeg. Firstly it's using floating point math while raspimjpeg sticks fairly rigidly to integer (with one exception down in [analyse_vectors2](https://github.com/roberttidey/userland/blob/master/host_applications/linux/apps/raspicam/RaspiMMotion.c)).
* Secondly your algorithm does expensive things like running pythagoras (squaring and importantly square roots) over the entire array. raspimjpeg doesn't bother with that; from a brief skim I think it's just summing the x and y vectors (which also doesn't require an expensive conversion to floats).
* Thirdly, you've got logging statements in your motion analysis; these are hugely expensive in Python, at least when they actually pass the logging filters. In order to ensure logs are definitely written and/or displayed when the code says they are, logging implicitly flushes its output - so if those logging statements actually output anything they'll have a pretty profound effect on performance.

Still, I'm quite impressed a single-core model like a B+ doesn't even max out when it's doing all this in a high level language like Python - try it on a 2B or a 3B sometime :)

waveform80, 2017-02-02:
Closing out old tickets; do feel free to re-open if you've still got a query about this!

closed 2017-02-02 by waveform80