waveform80/picamera #215: single recording file, circular buffer + video

eldonT, 2015-05-06:
Using the circular buffer with motion vectors detection, i'm trying to record to a single file when motion is detected, circular buffer + video recording for x seconds or until motion stops, but i can't find a way to do that properly using picamera functions.

Is there a way to do that keeping the stream consistency with picamera ?
Or do you have to postprocess the files afterwards like i'm trying to do, not very successfully, at the moment ?

thank you.

spikedrba, 2015-05-06:
@eldonT 
I could only get it done with postprocessing, here's the best script I was able to find/adapt, might be of inspiration:
https://gist.github.com/spikedrba/aecb82d8b51e991dbd01
By all means this doesn't mean that it's not possible to do it in memory, in fact you'd think that since all the data is there anyway you should be able to do that concatenation there and write to the SD only once, but haven't had a chance to figure that out.

best

eldonT, 2015-05-06:
thx, nice script

i am not using a background thread to merge my files, i was afraid it would produce i/o errors, but i like the idea.
Also i'm capturing at a higher "non regular" resolution (1600x1200) and i have already noticed dropped frames, but i think it is due to cpu/encoder limitations (and / or io problems).

I will take a look your full fov lower resolution with 2x2 binning. I was also not very pleased with the 1600x1200 resize quality either (aliasing), maybe using the zoom method would give better results (no resize) with a smaller fov but higher res.

eldonT, 2015-05-07:
something i didn't really think about.
am i right to assume the motion callback function is executed at each frame ?

i don't really need frame accurate detection (or 2 consecutive frames in spikedrba's script), so if that's the case i'll add a counter to perform detection only every x frames.

spikedrba, 2015-05-07:
@eldonT the custom output is called for every frame yeah

eldonT, 2015-05-10:
@spikedrba 
it seems i have a problem with your script, i can't change the video resolution to anything else than your original 1296x730.
when it is changed i get the following error, and the script hangs :

mmal: mmal_vc_port_enable: failed to enable port vc.ril.video_encode:in:0(I420): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.ril.video_encode:in:0(I420))0x23e0080 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled

Also, you're resizing a 16/9 aspect ratio to 4/3 for motion detection, is there a reason for that ?
i'll be using 4/3 capture resolution so i don't really care but i was wondering.

finally i don't quite understand what your histogram arrays are for, could you elaborate ? 
when i run you current script motion is almost constantly detected, i guess your sensitivity and thresholds are very low on purpose (?).

thanks

spikedrba, 2015-05-10:
@eldonT don't know what's up with the resolution issue, I just changed it to 2 different 4:3 modes to test (as listed here http://picamera.readthedocs.org/en/latest/fov.html) and has no problems with it. Are you positive the camera is not in use by a left over process or something by the time you try again with a different resolution?

No reason for the 16:9, that was just one of my initial tests, I now normally run in 4:3 as well, altho I'm not using tha whole script much anymore.

The histogram is just an explanatory mean to dump in the log so a human being can see what motion looked like. As far as high sensitivity, yes, that's basically cranked up to the maximum. As per comment right above the sensitivity setting (https://gist.github.com/spikedrba/aecb82d8b51e991dbd01#file-pmd-py-L93) if you wanted to catch 5% changes in your image at 640x480 you should use 61 as a valule.

Last but not least to give credit where it's due, this is really not my script, see the comment at the top, it was mostly crated by "killagreg" and you can find the original discussion here: http://www.raspberrypi.org/forums/viewtopic.php?p=656881#p656881

Also worth nothing someone else created a library to be used in python based motion projects here:
https://github.com/pageauc/pi-motion-lite

hope that helps,

Spike

eldonT, 2015-05-10:
thx for the comments
yes i can run raspivid in between script tests and have no problems whatsoever, so the camera module should be accessible by the script, it actually works fine as long as i don't touch the resolution value..

I'm wondering if some hidden char wouldn't have sneaked in there, i was also playing with serial usb terminal to manage my pi a+..
I will try to debug that later this week, i  was hoping it would work straight away, my lazy ass got bitten again, no worries.

thx for the link, i did begin to read (again) the original forum thread, i was actually using a modified version of pageauc first script (without vectors).

I also noticed this script loads the cpu quite heavily (20-30%), i'll see if processing motion only every x frames improves that a little.

thx again

waveform80, 2015-05-10:
It's pretty easy without post-processing. The key is to realize that the python idiom of "filename or file-like object" is extremely well thought out (and that picamera implements it *properly* - many libraries, sadly, do not). By this I mean that when you specify an output to, say, `split_recording` you have a couple of options:

1. You can specify a filename. This is the simple case ("just record to this file dammit!). In this case, picamera opens the file for you, and thus assumes responsibility for closing the file at the end of the recording (or capture).

2. You can specify a file-like object. This is the more complex case ("record to this specific object which looks like a file, at least as far as you know"). In this case, picamera just takes the object and calls its `write()` method now and again to write data to it. At the end of the recording/capture it does *not* close the file (because it never opened it), it merely calls the object's `flush()` method (if it has one) and leaves it for you to deal with.

The second case above makes it very easy to copy a ring buffer then carry on recording after it. Just open a file-like object, copy the ring buffer to it, then pass it as the target of `split_recording`. The library won't do anything silly like seeking to the start of the file (after all, it might be something unseekable, like a network socket) or even try and determine where it is in the file, it'll just keep recording to it, wherever it happens to be (at the end, as it happens).

I've thrown together a little gist (with lots of comments!) which demonstrates performing (crude) motion detection with H.264 motion vectors, and using a circular buffer which we split to/from when motion is/isn't detected, which can be found here:

https://gist.github.com/waveform80/9e990f56dabe27b42bfb

There's plenty of room for improvement in the script, but it should serve as a decent starting point for people to hack on!

spikedrba, 2015-05-10:
@eldonT weird, not really sure what's going on, the "script" is really just using picamera. Can you fire a python shell and run this successfully with whatever reasolution and framerate you're using?

```
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    camera.start_recording('my_video.h264')
    camera.wait_recording(60)
    camera.stop_recording()
```

regarding speed and cpu usage. Anything that's not using vectors and manipulating images for motion detection is basically useless imho since it happens CPU side and the ARM processor is just not fast enough for it. That said you're correct, it still takes a bunch of CPU and as far as I can see that's down to the vectors manipulation with numpy. Also see #213

waveform80, 2015-05-10:
Oh, the gist also has a nice recording symbol when it's recording (using overlays), so it relies on PIL/Pillow. I've tested it briefly while trying to get little-miss-screams-a-lot to sleep, but if you have any issues with it (like the PIL stuff not working), let me know and I'll try and expand the instructions a bit.

eldonT, 2015-05-12:
i hope i won't talk too much nonsense, it's a bit late and i realize i forgot about most of my perl.. 

So i believe there's a problem with motion_output and 1296x972 resolution, when either (i can't tell) you use a second splitter_port and/or motion resize of the main encoding (pageauc / spikedrba script)..

I can use spikedrba's script with various resolutions (1280x720, 1296x730) but with 1296x972 i get the mmal error message pasted above.
I haven't tried various resizing resolutions, i was worried about going from 16/9 to 4/3 aspect ratios but i don't know if it matters.. 

Now when i use waveform80's script which seems to use the same port as the main encoding, motion works with 1296x972..

As far as i can tell, the main difference between the 2 scripts is that one uses the main encoding to perform motion (one single start_recording) and the second uses a splitter_port "second" encoding with resizing to perform motion.

I tried to remove the re/size parameters but it didn't help.
I also noticed spikedrba's MotionDetector __init__ is missing, maybe something has to be set there.
At first i didn't understand why "size" was specified, i guess there's a MotionDetector class defined in picamera lib, i need to check that..

@spikedrba 
comparing your script with waveform80's, i get around 50% cpu load with yours and 25% with waveform80's.
Seeing the difference in the motion processing routines, my guess is most of that load difference comes from there. I yet have to test motion every only x frame.

My main concern is power draw and apparently that arm chip doesn't need much more power (10-15mA) to handle 50% load so..
Moreover i was using a Model B and now have a B+ and A+ (and pi2), so i already reduced power draw by 200-300mA, awesome :)

Also i don't know why but your script's merging (cat) of before-after parts, produces a file that makes my vlc crash (plays fine with mpv). I don't remember my script doing that and i am also using a simple cat (not threaded)..

@waveform80 
I had to add the following import to run your motion script :
import numpy as np

And thx a lot for python's "file like" capability, i will definitely check this out.

spikedrba, 2015-05-13:
@eldonT I think the difference in performances is down to two things:
- the script I'm using starts recording on two videopotrs using the splitter, @waveform80's only use one
- generating the histogram is probably taking some cycles, but that's just output, you could remove it, at which point the way we compute motion is the same

I'm recording on two video ports to use the resizer thinking that a lower res will take up less cpu for computation, but maybe i'm wrong there (it seems to be that way).

eldonT, 2015-05-14:
@spikedrba 
to answer my "bug" question above can you use 1296x972 resolution in your script or do you get the same error i have ?

yeah motion processing is important but i also failed to mention that waveform80's script was at 24fps not 40fps i'm using in your script..

I yet have to compare 2 splitters vs 1 cpu load, but also consider that with waveform80's script i am performing motion on full 1296x972, whereas in you script it's done on 640*480..
I guess the np math functions are fast enough so it doesn't really matter if the arrays are much larger.

Anyways i added a counter to perform motion detection only every x frames, i don't need to record events that are less than 250ms (every 10 frames @40fps = 4 detections / s), and with waveform80's script i get a steady 15% cpu load.

I also have to compare power draw for 2 vs 1 splitter, i already noticed that the higher the h264 resolution the higher the power draw, so my guess would be that 2 splitters will draw more than one single encoding..

@waveform80 
i have quickly tested your script linked above with the "file like" handle,  and it seems to work very well.
If you plan on adding CircularBuffer functions in your next version you should definitely include some picamera methods to offer that way of dumping the buffer and continue recording to the same file.

I was quite worried about the "cat before-after" merging command that was producing a very weird behavior on my pi B :
when "cat" is performed, all the network (rj45 plugged in) lights turn off for 2-3 seconds, then come back, ssh doesn't get disconnected and i don't get any errors in syslog, but that's really weird..
I already know that SD card I/O is problematic on the Pi, i was hoping that a class10 would help but it looks like the pi struggles with reading/writing large files (10+ MB)..

spikedrba, 2015-05-14:
@eldonT, yeah I had already tried, I have no problem with that resolution.

and yes, of course, twice as many fps will certainly make a huge difference in load. the double recording however will increase load too, I had sort of measured that myself, but it's also true that doing md on a much smaller resolution has an impact. That said I'm not sure about the cost of the resizer, maybe it performs the same doing it at full res as the additional computation is offset by no need for resizing.

eldonT, 2015-05-28:
i did some more testing and my mmal errors were most certainly related to GPU memory configuration. The higher the resolution, the more memory the encoder will require.

Some values may be enough to start the circular buffer with motion, but fail when capturing pictures while recording like i do.
Some lower values will simply send an Exception when running the first start_recording (motion or not).
As i am using A+ models (with ramfs), i started using limited GPU memory configurations.

I have not yet tested with one separate resized encoding for motion, as it is done in many scripts, but i expect it would require (much?) more memory.

My point is that mentioning minimum GPU memory required for resolutions encoding, would be a nice addition to picamera's documentation.

For example, with sensor mode=2, 1600x1200@15fps, single h264 encoder with motion, including capturing pictures while recording, you need a minimum of 80MB of GPU ram.

I'm also trying to locate the h264 hw encoder limitations, i'm not able to encode 1920x1440 at the moment, i remember reading that somewhere but can't find it back.

I am not entirely sure it's the only reason the python script failed to init the camera as i was able to run encodings at higher resolution with the same gpu config, using raspivid.
Is there a difference in the mmal encoder memory footprint between picamera and raspivid ?

I know there's also dynamic GPU memory allocation now on the rpi, could raspivid use it somehow or is it locked in boot/config, i'll check that later.