waveform80/picamera #217: Motion vectors recipe

spikedrba, 2015-05-12:
Hi Dave,

I'm trying to figure out a smarter algo for motion detection using the vectors and was playing around with your base recipe here:

http://picamera.readthedocs.org/en/latest/recipes2.html#recording-motion-vector-data

In the example where you manipulate the vectors the code reads:
```
width = 640
height = 480
cols = (width + 15) // 16
cols += 1
rows = (height + 15) // 16
```
why are you adding 15 and then doing a floordiv ? just in case people play around and set resolutoins that are not multiple of 16?

but more importantly, why the +1? what's the extra column for?

thanks,

spikedrba, 2015-05-12:
also do you have any thoughts why 15ps with wait_recording(10) would result in a len(motion_data) -> 143? what happened to the missing 7 frames? This is consistent across 3 consecutive recording tests (was collecting data at different levels of motion to compare the vectors).

waveform80, 2015-05-12:
The +15//16 is indeed to compensate for resolutions that don't precisely fit the macro-block size (16x16). As for the extra column ... good question! I've no idea exactly what it's for, but the firmware outputs it so I do too. I seem to recall various people asking about this on the camera forum but I don't think I've seen a definitive answer from the firmware devs yet.

As for missing frames that usually means that something's going too slow, so buffers aren't being returned fast enough, so frames get dropped (the claim in the docs that going too slow in the analysis classes causes things to crash is erroneous - it simply causes dropped frames - I must remember to fix that for the next release). What resolution are you using at 15fps? If it's something particularly huge that may explain the dropped frames; otherwise I'd have to dig into a profile-run to figure out where the dropped frames are coming from.

spikedrba, 2015-05-12:
thanks for the quick reply, makes sense. it's just 640x480 like in your example. Here's the capture script:

https://gist.github.com/spikedrba/6f4e9c264d5fadceb33d

waveform80, 2017-02-02:
Closing out some old tickets; I've just retried this with the latest firmware and the development version of picamera on a Pi3B and an old model B; both recorded 149 frames of motion data for each test scenario so whatever was causing the lost frames does seem to have been resolved. Do feel free to re-open this if it's still an issue!

closed 2017-02-02 by waveform80