waveform80/picamera #219: Silence capture

baptgauthier, 2015-06-03:
Hello,

Is it possible to capture without displaying the captured image ?

Thank
Baptiste

guyver2, 2015-06-03:
just don't start the preview.

```python
import time
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (1024, 768)
    # Camera warm-up time
    time.sleep(2)
    camera.capture('foo.jpg')
```

baptgauthier, 2015-06-03:
Thank for your response.
But I want the preview and take a capture after press a button.
This is impossible: preview + silence capture ?

Thank
Baptiste

guyver2, 2015-06-03:
using the video port :

```python
import time
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (1024, 768)
    camera.start_preview()
    # Camera warm-up time
    time.sleep(2)
    camera.capture('foo.jpg', use_video_port=True)
    time.sleep(2)
```

baptgauthier, 2015-06-03:
Perfect !
Thank

spikedrba, 2015-06-03:
in case it matters to you, worth keeping in mind that pictures taken from the video port will be "lower quality" than normal pictures as a bunch of filters are not applied. See http://picamera.readthedocs.org/en/latest/fov.html#the-video-port

baptgauthier, 2015-06-03:
Thank, 
but it is not possible to have the same quality?

Baptiste

spikedrba, 2015-06-03:
if you read that link you'll see that it's not a matter of quality per se, the original picture is the same, the point is postprocessing, which is why if you want to capture pictures fast you need to use the video port.

To be honest I don't know what your problem is to begin with, you originally asked about "capturing without displaying the captured image", and that's default behavior for me, but maybe because I don't run X?

can you please describe more in detail what your setup is and what you're trying to achieve? thanks

baptgauthier, 2015-06-03:
Basically, I want to do a photo booth.
So the person comes into a room with a screen (for preview) and a button to take a picture.
So I would like to have the best quality to keep photos.

You UNDERSTAND?
Thank

Sorry for my broken English, I'm french :(

spikedrba, 2015-06-03:
ok, I understand, thank for explaining (no need for CAPS). You definitely don't want to use the video port then, the image would be far worse looking otherwise. I still don't understand what your problem is, tho. If I open a python shell and use the snippet @guyver2 gave you (which btw comes straight from the documentation that I hope you've read already), I get "silence capture" as you called it.

What is it that you're actually seeing that's not what you want? please describe in details what happens step by step and what you want to happen.

thanks

baptgauthier, 2015-06-03:
Sorry pour la CAPS, it was not wanted.
I look at this in detail and makes you feedback.

Thank all

waveform80, 2015-06-03:
There's quite a lot that goes on when taking the picture with the still port instead of the video port. One of the reasons that the display changes when capturing the image (with the still port) is that the camera's mode is switched to sensor mode 2 or 3 (depending on selected framerate), for the duration of the capture. In other words, still port images are *always* captured at the full resolution of the camera (and then scaled down to whatever resolution the user requested). This is part of the reason for the higher quality (the other reason is a more powerful denoise algorithm applied to the results).

Naturally, if the preview is being run in another mode (for example, mode 5 if it's hooked up to a 720p TV) then the display will change (briefly) while the capture occurs. One way to minimize the display disruption (although it won't remove it completely) is to run the camera at full resolution all the time, but this does necessitate lowering the framerate to 15 (the maximum FPS the camera supports at 2592x1944):

```python
import time
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (2592, 1944)
    camera.framerate = 15
    camera.start_preview()
    # Warm up time
    time.sleep(2)
    camera.capture('foo.jpg')
```

I think that's the best you can do with picamera at the moment (assuming you need to stick with the still port). There's probably some lower level tricks for manipulating the preview I'm not aware of - might be worth asking about this in the Pi's camera forums.

baptgauthier, 2015-06-03:
Thank you, I'll test it :)

baptgauthier, 2015-06-06:
With the solution to down framerate, the preview is crappy.
I make : 

import time
import picamera
import RPi.GPIO as GPIO
import os

GPIO.setmode(GPIO.BCM)
GPIO.setup(23, GPIO.IN, GPIO.PUD_UP)

while True:
        with picamera.PiCamera() as camera:
                camera.start_preview()
                print "Wait entry"
                GPIO.wait_for_edge(23, GPIO.FALLING)
                print "Take picture"
                os.system("su pi -c /home/pi/scripts/bash/sourire.sh")
                file_name = '/home/pi/scripts/images/sample_%s.jpg'% str(time.time())
                camera.capture(file_name,use_video_port=True, format='jpeg', quality=100, thumbnail=None)
                os.system("su pi -c /home/pi/scripts/bash/last.sh")
                time.sleep(1)
                camera.close()
                time.sleep(3)
                os.system("su pi -c /home/pi/scripts/bash/merci.sh")

With script bash, i change background desktop when camera stopping and display current capture.
the script works correctly but the quality is medium

thank

waveform80, 2015-06-27:
If by "crappy" you mean that the preview is considerably less "smooth" than it is at higher framerates (>24) then I'm afraid that's the trade-off you have to make. If you're using the still port to capture images (`use_video_port=False`, the default) then the camera *will* switch to the max. res 15fps mode for the duration of the capture causing the "display" of the image you originally mentioned. Alternatively you can run in the 15fps mode permanently in which case the mode switch is unnecessary, but you have a slightly "jerky" preview. If you want anything more sophisticated than that it'll need to be done at the firmware level (assuming it's possible), but that's beyond my capability.

closed 2016-05-07 by waveform80