waveform80/picamera #221: Same color and levels for various pictures

starcraft04, 2015-06-17:
Hi,

I am trying to take pictures every hours of some objects and I need to have the exact same parameters in order to have the same levels and colors for every picture. I followed the guide 4.6. Capturing consistent images in the documentation and it is working but not 100%, I still can see, even if subtle, some differences in colors. For example the white part of the object is sometimes a little bit more yellow in some cases.

I was wondering if there was a way to get the exact same levels, balance and colours for all the pictures knowing that the light is not changing and the object is only moving from one position to another.

Thanks,

John.

waveform80, 2015-06-27:
Hi John, is there any chance you could post a code example? I'm concerned that you may be re-initializing the camera in between captures. In this case the levels won't be the same as the auto-exposure and auto-white-balance algorithms will have adjusted to different scenes. The only way to guarantee similar exposures is to keep the camera running with the same settings between captures.

With hours between exposures that may seem strange but the problem is that the firmware provides no way of controlling the camera gains - once can either let them float according to the AGC algorithm, or one can fix them at their current levels (by contrast, the AWB levels can be specified explicitly).

waveform80, 2016-05-07:
@starcraft04 - just cleaning up old issues and questions - did you still need any help with this one?

closed 2017-02-02 by waveform80