waveform80/picamera #222: Any plans to support Win10 IoT Core?

elmerm, 2015-06-21:
Just want to know if you're planning to port/support Win10 IoT Core at all or if that's something Microsoft is doing. Would love to use this on Win10. Please advise.

waveform80, 2015-06-27:
That's a very interesting question. Personally I'd like to support picamera on Win10 IoT, but bear in mind that picamera is basically a thin wrapper over the libmmal library (which in turn is an API to the camera firmware). In other words, the scenarios break down like this:

1. Camera firmware is ported and libmmal is ported too: easy peasy, I can probably port picamera over with minimal effort and I'd be happy to do so.

2. Camera firmware is ported, but there's a different userspace library. In this case porting picamera is probably possible but there'll be a serious amount of effort involved and it'll largely depend on either my free time or someone stepping up to help.

3. Camera firmware isn't ported: no chance.

I suspect whether the camera firmware and/or libmmal is ported, is up to the foundation (or more specifically the (ex-)Broadcom people that maintain said firmware); and I should stress at this point that I don't work for the foundation, hence I know nothing about their plans in this area!

Or to put it another way: @bennuttall any ideas what's happening w.r.t Win10 and the camera? (that you can discuss publicly!)

bennuttall, 2015-06-28:
I know nothing about Win10. See what @ghollingworth can tell you?

ghollingworth, 2015-06-28:
I cannot pre-empt what Microsoft are working on but they were talking about integrating the video functionality.

To be clear, Raspberry Pi ourselves are not supporting the Windows platform, we've got enough on our plates with the Raspbian distribution!

Gordon

waveform80, 2015-06-29:
Many thanks Gordon & Ben! Is there anyone we could ask at Microsoft to drop by and comment on their plans at this stage? Also by "video functionality" do you happen to mean the video recording capability or just video playback?

GuardRex, 2015-06-29:
I know that Scott Hanselman @shanselman and Damian Edwards @DamianEdwards have been hacking around with the Pi, Win10, and .NET. They mention it from time-to-time on the .NET Community Standup (and make it clear that they are just hacking around with no official plans by MS to explicitly release a version of Win10 for the Pi anytime soon).

elmerm, 2015-06-29:
@GuardRex Not sure if you know but there is an official version of Win10 for the Pi from MSFT. See http://ms-iot.github.io/content/en-US/win10/SetupRPI.htm. It's a preview release currently.

GuardRex, 2015-06-29:
Ah! I must have been thinking about .NET 5 Core CLR. When the guys mention this subject on the .NET 5 Community Standup and they talk about "hacking it" and "not officially supporting it," they must mean .NET 5 Core CLR running on the Pi. I stand corrected on the Win10 part. Thanks.

waveform80, 2016-05-07:
I'll close this for now unless - but if anyone's got any interesting updates on Win10 IoT core, do feel free to post them. Unfortunately between picamera, gpiozero, picraft, etc. my time for open-source projects is basically saturated so I can't say I'll be working on this in the foreseeable future.

closed 2016-05-07 by waveform80