waveform80/picamera #223: how to reduce capture time

prer2020, 2015-06-26:
Hello everyone.

For my project application I need very less time around 2 ms to capture the image. Is it possible to capture an image in 2ms ? 

I have written simple program:
with picamera.PiCamera() as camera:
camera.resolution = (640, 480)
camera.framerate = 90
camera.video_stabilization = True
camera.shutter_speed = 800
camera.start_preview()
time.sleep(1)
start_time = time.time()
camera.capture('/home/pi/30th may/img14.jpg','jpeg',True)
print (time.time()-start_time)

I want to reduce capturing time. Here I am getting time 190ms but practically speaking it should take 11ms for each frame. whenever i am capturing 90 frames sequentially it takes approximately 1 sec means each frame it taking 11ms but if i am writing the program to capture single frame it is taking more than 100ms . what are the options to reduce to this capturing time

6by9, 2015-06-26:
History from https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=113971
The issue appears to be setup time, as continuous capture gives the expected 11ms between frames. Is the preview port set up with the same framerate parameters as capture? If it is dropping back to the 30fps mode after each capture, that would explain it.

prer2020, 2015-06-26:
I have not set FPS for preview port. How to set FPS for preview port?

6by9, 2015-06-26:
@prer2020 My comment was aimed at waveform80 rather than yourself as. You hadn't provided the information from the forum thread and I wanted to save him having to ask the same questions again.

waveform80, 2015-07-07:
@6by9 - thanks for the forum pointer!

@prer2020 - 6by9's absolutely right. By far the biggest cost in your script is the time it takes picamera to initialize. For example here's a hacky little script that outputs various timings:

```python
import picamera
import time

def how_long(start, op):
    print('%s took %.2fs' % (op, time.time() - start))
    return time.time()

start = time.time()
with picamera.PiCamera() as camera:
    start = how_long(start, 'init')
    camera.resolution = (640, 480)
    start = how_long(start, 'set resolution')
    camera.framerate = 90
    start = how_long(start, 'set framerate')
    camera.video_stabilization = True
    start = how_long(start, 'set video_stabilization')
    camera.shutter_speed = 800
    start = how_long(start, 'set shutter_speed')
    camera.start_preview()
    start = how_long(start, 'start_preview')
    time.sleep(1)
    start = time.time()
    camera.capture('img14.jpg', 'jpeg', use_video_port=True)
    start = how_long(start, 'capture')
```

On my Pi 2 this gives:

```
init took 0.25s
set resolution took 0.23s
set framerate took 0.24s
set video_stabilization took 0.00s
set shutter_speed took 0.00s
start_preview took 0.04s
capture took 0.13s
```

So init takes about a quarter of a second, as does setting resolution and framerate (this isn't too surprising - setting the resolution or the framerate causes picamera to re-init the camera hence why the timings are so similar). Everything else is trivially fast until we get to the capture which only takes 0.13s (that's including writing out the file as well).

Assuming you want to capture an image quickly in response to something you want to avoid that init time by starting up the camera and waiting on some event to tell you when to capture. There's an example of doing this in response to a GPIO button push in the worksheet at https://www.raspberrypi.org/learning/python-picamera-setup/worksheet.md (look for the heading "Camera programming: capture when activated")

waveform80, 2015-07-07:
Incidentally, if you want to speed up resolution and framerate setting, picamera now permits initial values of these to be set in the constructor (I think that was in 1.9 or 1.10 - I forget which). Anyway, that means you can speed up the script considerably like so:

```python
import picamera                                                                 
import time                                                                     
                                                                                
def how_long(start, op):                                                        
    print('%s took %.2fs' % (op, time.time() - start))                          
    return time.time()                                                          
                                                                                
start = time.time()                                                             
with picamera.PiCamera(resolution=(640, 480), framerate=90) as camera:          
    start = how_long(start, 'init')                                             
    camera.video_stabilization = True                                           
    start = how_long(start, 'set video_stabilization')                          
    camera.shutter_speed = 800                                                  
    start = how_long(start, 'set shutter_speed')                                
    camera.start_preview()                                                      
    start = how_long(start, 'start_preview')                                    
    time.sleep(1)                                                               
    start = time.time()                                                         
    camera.capture('img14.jpg', 'jpeg', use_video_port=True)                    
    start = how_long(start, 'capture')
```

Which now gives me:

```
init took 0.25s
set video_stabilization took 0.00s
set shutter_speed took 0.00s
start_preview took 0.07s
capture took 0.12s
```

(don't pay much attention to the difference in the capture time there - there's enough disk IO going on with a capture to make that time very variable)

closed 2016-05-07 by waveform80

sleebapaul, 2018-09-27:
Anyone knows how to set the resolution of PiCamera? Say `640x480`?