waveform80/picamera #226: stream UDP

Flojou, 2015-07-01:
Hello I would like to send the stream from the camera in real time with the UDP protocol .
Sorry for my English and thank you in advance.

waveform80, 2015-07-07:
Sure - just modify the ["Recording to a network stream"](http://picamera.readthedocs.org/en/release-1.10/recipes1.html#recording-to-a-network-stream) recipe to use a UDP socket instead of a TCP one. It's a little tricky given that UDP is connectionless so you can't just listen and accept a connection, instead you basically process every incoming packet and throw it at the player (in the vague hope that nothing's been dropped, repeated, re-ordered, etc.) but it's not hugely difficult. The server side would look something like this:

```python
import socket
import subprocess

# Start a socket listening for connections on 0.0.0.0:8000 (0.0.0.0 means
# all interfaces)
server_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
server_socket.bind(('0.0.0.0', 8000))

# Accept a single connection and make a file-like object out of it
try:
    # Run a viewer with an appropriate command line. Uncomment the mplayer
    # version if you would prefer to use mplayer instead of VLC
    cmdline = ['vlc', '--demux', 'h264', '-']
    #cmdline = ['mplayer', '-fps', '25', '-cache', '1024', '-']
    player = subprocess.Popen(cmdline, stdin=subprocess.PIPE)
    while True:
        # Repeatedly read up to 128k of data from the connection and write it
        # to the media player's stdin; we deliberately use a large size here to
        # avoid truncating UDP messages
        data = server_socket.recv(131072)
        if not data:
            break
        player.stdin.write(data)
finally:
    connection.close()
    server_socket.close()
    player.terminate()
```

And the Pi side looks something like this:

```python
import socket
import time
import picamera

# Connect a client socket to my_server:8000 (change my_server to the
# hostname of your server)
client_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
client_socket.connect(('my_server', 8000))

# Make a file-like object out of the connection
connection = client_socket.makefile('wb')
try:
    with picamera.PiCamera() as camera:
        camera.resolution = (640, 480)
        camera.framerate = 24
        # Start a preview and let the camera warm up for 2 seconds
        camera.start_preview()
        time.sleep(2)
        # Start recording, sending the output to the connection for 60
        # seconds, then stop
        camera.start_recording(connection, format='h264')
        camera.wait_recording(60)
        camera.stop_recording()
finally:
    connection.close()
    client_socket.close()
```

You'll probably find a fair amount of frame corruption from dropped / re-ordered packets but, hey, that's what you get for using a datagram protocol :)

Flojou, 2015-07-09:
thank you for your response . But I have an error in the client program (PI) :

pi@raspberrypi ~ $ python test.py 
Traceback (most recent call last):
  File "test.py", line 22, in <module>
    connection.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
socket.error: [Errno 89] Destination address required

Thank you in advance for your answer.

waveform80, 2015-07-09:
Did you change `'my_server'` to the hostname/ip-address of the machine running the server script?

Flojou, 2015-07-15:
Yes, I Replaced I' my_server ' by my IP address.
I can find another way to send the stream UDP with VLC like this:
 Popen ( [' cvlc ', ' stream' , video , '--sout' ,'#standard{access=udp,dst=239.255.1.1 :6000}' ] )
the problem is how to get the flow on a variable (video = ????)

Flojou, 2015-07-22:
My problem is solved but I like to know if you can read the stream received from direcement VLC ( vlc udp://@:8000) Thanks

waveform80, 2015-07-22:
The H264 NAL stream output by the Pi's camera is not encapsulated in a nice container like MP4. Because of this VLC doesn't know how to play the output unless you explicitly tell it that it's H264 NAL. You don't need to do this when you're just playing a file because VLC guesses it from the filename extension, but in the case of a network stream it doesn't have that option. So that's why there's a "--demux h264" option added on the server side. Try it with the following command line:

```
vlc udp://@:8000 --demux h264
```

That works for me.

Flojou, 2015-07-22:
Thank you it works well. One last question and you know how to do multicast (ip = 239.1.2.3:50000) still udp.I find this:
   sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
   sock.setsockopt(socket.IPPROTO_IP, socket.IP_MULTICAST_TTL, 2)

but it does not work

waveform80, 2015-07-22:
That's a good question - I've never worked with multicast myself, just good old fashioned broadcast (which compoundpi uses) so I'm afraid I can't help much there. After a quick google I did run across this on SO: http://stackoverflow.com/questions/603852/multicast-in-python - and from that I suspect you need to do more than merely set the socket options for multicast (and I'd probably use a TTL a little longer than 2 just in case there's anything that counts as a router in the way). Have a look at the `sendto` calls in the answers. In each case they're using `sendto(buf, (MCAST_GRP, MCAST_PORT))` so I'm guessing you'd need a little more than a file-wrapped socket. Still, shouldn't be too hard to use a [custom output](http://picamera.readthedocs.org/en/release-1.10/recipes2.html#custom-outputs) to achieve this. Something like:

```python
import socket
import picamera
#...

class MyOutput(object):
    def __init__(self, socket):
        self.socket = socket

    def write(self, buf):
        # Need to define MCAST_GRP and MCAST_PORT somewhere
        self.socket.sendto(buf, (MCAST_GRP, MCAST_PORT))

#...
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_MULTICAST_TTL, 32)
#...
camera.start_recording(MyOutput(sock), format='h264')
#...
```

Flojou, 2015-07-23:
Your solution works , thank you .
I also succeeded with your first solution.
Thank you for your help, you helped me a lot

rmackay9, 2016-02-24:
I am also trying to use the picamera's start_recording feature to stream the RPI2 camera's video stream to an Ubuntu machine using UDP.  What I've found though is that on the Ubuntu side it's complaining that it has not received a valid RTP payload.

On the sending side (an RPI2) I've used code identical to what's suggested by @waveform80 above (except I've changed the IP address accordingly).

On the Ubuntu machine I'm running this:
gst-launch-1.0 -v udpsrc port=9000 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! videoconvert ! ximagesink

On the Ubuntu side I'm getting this error message:
gstrtpbasedepayload.c(381): gs_rtp_base_depayload_chain (): /GstPipline:pipeline0/GstRtpH264Depay:rtph264depay0:
**Received invalid RTP payload, dropping**

I guess maybe the method described above does not send some important info that the receiver needs.  I guess it works RPI->RPI but isn't compatible with gstreamer?  .. or perhaps I need some change to the streamer command line.

Anyway, I certainly like picamera.

waveform80, 2016-02-24:
The code above doesn't generate an RTP stream; it just dumps the H264 NAL packets onto the network over UDP. You'd need to do a lot more work to encapsulate the stream in [RTP](https://en.wikipedia.org/wiki/Real-time_Transport_Protocol) (or get something like gstreamer to do the work for you).

closed 2016-05-07 by waveform80

houssammoncef, 2017-04-12:
hello i want to change the part of vlc with cv2 or other stuff to display it in gui
and thanks for your advices