waveform80/picamera #227: framerate() setting disables preview and zoom, causes long capture delay

plsherman, 2015-07-04:
I need to use the Pi camera for long exposure images. The default intialization of the camera only allows a maximum 33ms shutter speed. The framerate() call will allow slower shutter speeds but causes three unwanted side effects:
1. There's no preview image
2. Image capture takes almost 30 seconds
3. The zoom() function is disabled 
I'm using the following code to run the camera. If I remove the framerate() call, the camera functions normally, with a preview and a fraction of a second to capture the image. The zoom call is commented out because it does nothing and I can use the image without it. When the capture() function is invoked, there's almost a 30 second delay then a short flash of the image on the display just before the capture() function completes.

my_file = 'test.jpg'
with picamera.PiCamera() as camera:
    camera.resolution = (2582,1944)
    camera.framerate = Fraction(1,6)
#    camera.zoom = (0.25,0.25, 0.25,0.25)
    camera.shutter_speed =190000
    camera.iso = 800
    camera.start_preview()
    time.sleep(5)
    camera.capture(my_file)

waveform80, 2015-07-07:
The basic problem is that you're using an incredibly slow framerate without waiting long enough for it! Firstly a bit of background: when you capture a frame with the still port (when `use_video_port=False` or omitted) a lot of stuff happens in the background:

1. There's a mode switch to the highest resolution (2592x1944)
2. A frame is retrieved and discarded (as the first frame returned by the camera is typically corrupt after a mode switch)
3. Another frame is retrieved and processed
4. The mode is switch back to the original

(I should point out I'm no expert on the firmware - the above is a rough sketch based on what I've read from the firmware devs on the forums)

Now consider that with a framerate of 1/6th fps there will be a gap of 6 seconds between each frame, so each step above will take 6 seconds (in order to avoid doing things in the middle of a frame). The reason there doesn't appear to be a preview is simply that you're not waiting long enough for it. Change your delay from 5 seconds to 60 and you'll see the preview (but of course that'll increase the runtime even further :). Initially it'll appear black because the AGC gains are initially 0, and AGC usually requires a few frames to measure exposure correctly (and remember each frame is taking 6 seconds so a few frames rapidly adds up to a lot of time). It'll also appear horrendously green (once it starts appearing) because AWB needs some time to figure out a decent balance once AGC has figured out decent gains.

Incidentally, the zoom seems to work fine when you wait a bit longer.

However, it's worth noting you're only using a shutter speed of 190,000µs (0.19s) You can achieve this with a much faster framerate of 5fps (which will produce a frame every 0.2 seconds) which should enable everything to work more "normally" without huge pauses:

```python
import picamera
import time
from fractions import Fraction

my_file = 'test.jpg'
with picamera.PiCamera() as camera:
    camera.resolution = camera.MAX_RESOLUTION
    camera.framerate = 5
    camera.zoom = (0.25, 0.25, 0.25, 0.25)
    camera.shutter_speed = 190000
    camera.iso = 800
    camera.start_preview()
    time.sleep(5)
    camera.capture(my_file)
```

plsherman, 2015-07-08:
Thanks for the reply explaining what's happening. Unfortunately, at the
late hour I posted the question, I forgot to change the exposure time
back to the time I'm using for the actual exposure; around 1 second. I
can still speed things back up quite a bit by correctly setting the
frame rate. I'll also be disabling AWB and, if possible, AGC (exposure
mode 'off') which will hopefully help improve performance.

Phil

closed 2016-05-07 by waveform80