waveform80/picamera #228: Accessing framerate as a tuple is deprecated; this value is

bootsmann, 2015-07-06:
Why I get this warning...
```/usr/lib/python2.7/dist-packages/picamera/camera.py:151: PiCameraDeprecated: Accessing framerate as a tuple is deprecated; this value is now a Fraction, so you can query the numerator and denominator properties directly, convert to an int or float, or perform arithmetic operations and comparisons directly
  'Accessing framerate as a tuple is deprecated; this value is '```

...with following code?
```python
with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    camera.hflip = True
    camera.vflip = True
    camera.start_preview()
    time.sleep(2)```

waveform80, 2015-07-06:
I don't get that warning with that code (unsurprisingly given it's not touching `framerate`), at least when using the latest version 1.10. You can check the version you've got installed with the instructions here:

http://picamera.readthedocs.org/en/release-1.10/faq.html#how-can-i-tell-what-version-of-picamera-i-have-installed

However, I suspect there's more to your code that you haven't pasted?

bootsmann, 2015-07-07:
```python
from pkg_resources import require
require('picamera')
[picamera 1.10 (/usr/local/lib/python2.7/dist-packages)]
```

Entire code:
https://github.com/miguelgrinberg/flask-video-streaming/blob/master/camera_pi.py

waveform80, 2015-07-07:
Thanks for the info - very helpful stuff. Unfortunately I still can't reproduce the deprecation warning; here's the steps I used, just so we can compare notes. On my Pi 2 I set up a virtual env for the project, cloned your repo, installed from requirements-pi.txt, modified app.py to use camera_pi.py instead of camera.py, and added a line to enable deprecation warnings, then ran the server:

```
dave@gonzo ~ $ git clone https://github.com/miguelgrinberg/flask-video-streaming.git
Cloning into 'flask-video-streaming'...
remote: Counting objects: 23, done.
remote: Total 23 (delta 0), reused 0 (delta 0), pack-reused 23
Unpacking objects: 100% (23/23), done.
dave@gonzo ~ $ virtualenv flaskstreamenv
New python executable in flaskstreamenv/bin/python
Installing distribute..................................................................................................
...........................................................................................done.
Installing pip...............done.
dave@gonzo ~ $ source flaskstreamenv/bin/activate
(flaskstreamenv)dave@gonzo ~ $ cd flask-video-streaming/
(flaskstreamenv)dave@gonzo ~/flask-video-streaming $ pip install -r requirements-pi.txt 
Downloading/unpacking Flask==0.10.1 (from -r requirements-pi.txt (line 1))
  Downloading Flask-0.10.1.tar.gz (544Kb): 544Kb downloaded
  Running setup.py egg_info for package Flask
    
    warning: no files found matching '*' under directory 'tests'
    warning: no previously-included files matching '*.pyc' found under directory 'docs'
    warning: no previously-included files matching '*.pyo' found under directory 'docs'
    warning: no previously-included files matching '*.pyc' found under directory 'tests'
    warning: no previously-included files matching '*.pyo' found under directory 'tests'
    warning: no previously-included files matching '*.pyc' found under directory 'examples'
    warning: no previously-included files matching '*.pyo' found under directory 'examples'
...
[snipped for brevity]
...
(flaskstreamenv)dave@gonzo ~/flask-video-streaming $ vim app.py 
(flaskstreamenv)dave@gonzo ~/flask-video-streaming $ python app.py                                                     
 * Running on http://0.0.0.0:5000/
 * Restarting with reloader
192.168.80.157 - - [07/Jul/2015 11:04:57] "GET / HTTP/1.1" 200 -
/home/dave/flaskstreamenv/local/lib/python2.7/site-packages/picamera/camera.py:2488: PiCameraDeprecated: PiCamera.ISO i
s deprecated; use PiCamera.iso instead
  'PiCamera.ISO is deprecated; use PiCamera.iso instead'))
192.168.80.157 - - [07/Jul/2015 11:05:00] "GET /video_feed HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.80.157', 47423)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 651, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 704, in finish
    self.wfile.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------
```

Now, as you can see in the output above, I did pick up a deprecation warning but a different one to the one you reported. However, I recognized that one as being something that occurred on camera init in version 1.8 (#163) which in turn led me to notice that requirements-pi.txt is actually installing picamera-1.8 in the venv instead of 1.10:

```
^C(flaskstreamenv)dave@gonzo ~/flask-video-streaming $ cat requirements-pi.txt 
Flask==0.10.1
Jinja2==2.7.3
MarkupSafe==0.23
Werkzeug==0.9.6
itsdangerous==0.24
picamera==1.8
```

Just replace that with picamera==1.10 (or better still picamera<2.0dev) and that deprecation warning should disappear.

However, that still leaves us with the question of the deprecation warning you originally reported. As you can see, that's not appearing for me - although I'm not terribly surprised. That warning should only appear if you access the framerate property and attempt to treat the return value as a tuple (it's actually a Fraction subclass nowadays so you can just treat it as you would any floating point value). For an example, see the [relevant section in the deprecation chapter of the docs](http://picamera.readthedocs.org/en/release-1.10/deprecated.html#fractions-as-tuples). 

I can't find any such usage in your code though, so you must be working from a different version than that which is in your GitHub repo (I had a look at prior commits of camera_pi.py in there but didn't find anything).

bootsmann, 2015-07-07:
I work with bottle.
http://pastebin.com/Fsaz17U9

```
Bottle v0.12.8 server starting up (using WSGIRefServer())...
Listening on http://10.0.2.114:8080/
Hit Ctrl-C to quit.

/usr/local/lib/python2.7/dist-packages/picamera/camera.py:151: PiCameraDeprecated: Accessing framerate as a tuple is deprecated; this value is now a Fraction, so you can query the numerator and denominator properties directly, convert to an int or float, or perform arithmetic operations and comparisons directly
  'Accessing framerate as a tuple is deprecated; this value is '
Traceback (most recent call last):
  File "/usr/lib/python2.7/wsgiref/handlers.py", line 86, in run
    self.finish_response()
  File "/usr/lib/python2.7/wsgiref/handlers.py", line 127, in finish_response
    self.write(data)
  File "/usr/lib/python2.7/wsgiref/handlers.py", line 215, in write
    self._write(data)
  File "/usr/lib/python2.7/socket.py", line 324, in write
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------
Exception happened during processing of request from ('10.0.2.112', 55274)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 651, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 704, in finish
    self.wfile.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------
^CTraceback (most recent call last):
  File "/usr/lib/python2.7/wsgiref/handlers.py", line 86, in run
    self.finish_response()
  File "/usr/lib/python2.7/wsgiref/handlers.py", line 126, in finish_response
    for data in self.result:
  File "app.py", line 20, in gen
    time.sleep(.1)
KeyboardInterrupt
root@rpi:/home/pi/Python/Bottle# Camera closed
```

Usually I use CherryPy or gevent as a webserver and so I get no socket error with multiple simultaneous connections

waveform80, 2015-07-07:
I don't think the choice of web-server is going to change anything; the point is that camera_pi.py (at least the version that's at the head of your repo) isn't accessing framerate as a tuple so there's no way I can reproduce that deprecation warning. Whatever variant of camera_pi.py you've got, it isn't the same as the one at the head of your repo.

bootsmann, 2015-07-07:
camera.py
http://pastebin.com/t6x14eEF

Now I am rather at a loss

waveform80, 2015-07-07:
Aha! Now I can reproduce the issue. I'm still a bit stumped as to exactly where the issue is being caused but at least I can reproduce it, so I stand a chance of figuring it out :) Should have something soon...

waveform80, 2015-07-07:
Right, this is definitely a bug in picamera 1.10; `_set_camera_mode` uses the deprecated method of accessing the framerate. I'll get this fixed for 1.11.

I also figured out why I couldn't reproduce it: the issue doesn't exist in picamera 1.8 (specified by requirements-pi.txt) but was introduced prior to picamera 1.10. I'd upgraded the picamera installation in the virtualenv but forgotten to actually test it, hence why I was suddenly able to reproduce the issue with your paste-binned code. Your repo master reproduces the issue as well, but only when used with picamera 1.10.

bootsmann, 2015-07-07:
Great job Dave. Thx for support

waveform80, 2015-07-07:
No prob; don't close this yet by the way - I'll close it with the commit when I've fixed it (that won't be today unfortunately - too busy with a couple of other projects, but I should get to it shortly :)

waveform80, 2016-03-17:
Shortly, hah! Well, that's what happens when you have kids, folks. Anyway, time to get this sorted...

commit bd555c6d (2016-03-17): Fix #228

Ensure setting resolution doesn't raise deprecation warnings
closed 2016-03-17 by waveform80