waveform80/picamera #230: Camera Issue When No Display Present

tsloire, 2015-07-09:
I have read the documentation but cannot find any reference to my issue, so I apologise in advance if I have missed something!

I have got picamera working great whilst I have both a hdmi monitor, usb keyboard and usb mouse (both t through a powered usb hub) plugged into my Raspberry Pi B.  I am using a slightly different camera module than the normal one, which is this one:  http://www.ebay.co.uk/itm/5MP-160-Degree-Wide-Angle-Fisheye-Raspberry-Pi-Camera-Module-Board-High-Quality-/231580210964?pt=LH_DefaultDomain_3&hash=item35eb419714

Like I said though that camera works when I use it with everything plugged in.  However, when I boot up my pi and run the program automatically without any of the peripherals (barring the camera) plugged in, the camera will take photos but the photos will have no data in them and not present me with a picture.  If I then plug the hdmi back in and take a photo without rebooting the pi, the camera takes photos fine.

This is kind of a weird issue, maybe I am missing something, I'm not sure, but I'm going to test this out with the normal camera too and see if the same thing happens.  Until then, do you have any advice on how to fix this issue or what I might be doing wrong?

Cheers for any help!
TSLoire

waveform80, 2015-07-09:
Hmm ... curious. When no display is detected, picamera should default to using the 720p resolution. In fact my main development pi (currently an old rev2 Model B) always runs headless with only the camera module and ethernet connected to it (I test preview related stuff on a separate Pi 2 connected to a display). Could you try running the following script when nothing is plugged in and let me know the output in `resolution.txt`?

```python
import io
import picamera

with picamera.PiCamera() as camera:
    with io.open('resolution.txt', 'wb') as f:
        f.write(b'Resolution: %dx%d\n' % camera.resolution)
```

That should let me know if something weird is going on with the resolution selection. Beyond that though I'd probably have to have a look at your script to figure out what's going on.

tsloire, 2015-07-10:
I'll give that script a run when I get back home on Monday, I'm out all weekend now.  I'm also running it on battery power, but the RPi still works fine when the monitor and everything else is plugged in.

I'll also post my script up on here so you can see what I'm doing.  No doubt it's bad!

Cheers for the quick response!

tsloire, 2015-07-18:
Hey Waveform, sorry it has taken so long to get back to you on this, I've been very busy with other things this past week (girlfriend moving in!).

I ran that code and got the following output:  'Resolution: 640x480'

So it didn't default to 720p  I'm running it so that the PI boots up and loads my script in the terminal so I can quickly plug in a display if needs be and monitor it.  As well as that, I am using pins GPIO 17 and the ground pin next to it.  Could either of these facts be causing trouble?

I have attached my script below with notes to show what I am doing.. which is probably wrong :P  I have tried using 'with PICam.PiCamera() as camera:' before but had the same issue with that too, the benefit of that is that the camera is not constantly running as it is in the script I have pasted below.  In my first script it waited until pin 17 went LOW then took a photo.  Anyway, here is my script:

#!/usr/bin/python

import RPi.GPIO as GPIO
import time
import picamera
import os
import os.path

GPIO.setmode (GPIO.BCM)
GPIO.setup (17, GPIO.IN, GPIO.PUD_UP)

camera = picamera.PiCamera()
buttonPressed = False

def camera_TakePhoto ():
    
    ## DO I have to use this?  Will it break otherwise?
##  with PICam.PiCamera() as camera:

    ## Set the default photo number here
    photoNumber = 001
    photoNumberStr = ''
    fileName = ''
    fileMatchCount = 0

    ## Cycle through a list of files whose names contain the phrase 'RPiCam_LE_' and grab the number at the end of them
    ## Set the current photo number to the last image's number +1 so no overwriting occurs
    for file in os.listdir ('/home/pi/Desktop'):
        if file.startswith ('RPiCam_LE_'):
            fileMatchCount += 1
            splitStr = file.split ('_')
            splitStr = splitStr[2].split ('.')
            splitStr = splitStr[0]
            if int(splitStr) >= photoNumber:
                photoNumber = int(splitStr) + 1
                photoNumberStr = str (photoNumber)

    ## If no files exist with the phrase 'RPiCam_LE_' then the photo number is 001                
    if fileMatchCount == 0:
        photoNumber = 001
        photoNumberStr = str (photoNumber)
        
    ## Tell the camera to  capture an image and save it to the specified location
    camera.capture ('/home/pi/Desktop/RPiCam_LE_' + photoNumberStr + '.jpg')

    print ('Done taking photo!')
    

while True:

    ## Grab the stage of GPIO 17
    input_state = GPIO.input (17)

    ## If the pin is HIGH and the button isn't currently pressed, set the button pressed bool to true and run the camera_TakePhoto function and wait
    if input_state == True and buttonPressed == False:
        buttonPressed = True
        camera_TakePhoto ()
        time.sleep (0.2)

    ## If the pin is LOW and the button is currently pressed, set the button pressed bool to false
    elif input_state == False and buttonPressed == True:
        buttonPressed = False
        time.sleep (0.2)

Any help would be great!  I am going to try using the normal PiCam too instead of my 160 degree camera to see if that makes a difference.  Cheers Waveform, hopefully you can help solve my mystery!

tsloire, 2015-07-18:
Oh man, I have done something wrong pasting that code in, it's gone bonkers, sorry I'm not used to posting on github.  Everything from 'here is my script:' to 'Any help would be great!' is my code.  FOr some reason I cannot edit my comment on my Pi either =/

waveform80, 2016-05-07:
@tsloire sorry it's taken me so long to get back to this! Real life and all that...

There's nothing obviously wrong with your code (and you're fine to use or not use the "with picamera..." construct instead of "camera = picamera.PiCamera()" - either's fine). If your completely headless Pi is capturing at 640x480 that suggests the boot config is telling the display to operate at that res - do you have any `hdmi_group` / `hdmi_mode` lines in your `/boot/config.txt`? If not, the Pi may just be using that as a default res in the case of no attached display.

The 1280x720 res is used when the Pi has no active display at all - I should've been clearer about what this means: it basically only applies when `tvservice -o` has been run which disables the display entirely. In this case picamera can't detect any display (even if one's physically connected) so I figured 720p was a "nice" default.

As to why your images are coming out black ... I'm not sure, but it sounds like something odd with your hardware (picamera's definitely happy running headless - it's a pretty common use case for it).

Anyway, I've just been going through the backlog of tickets trying to clear up stuff - I'll leave this open for now in case you still need a hand with this - sorry again about the long delay!

waveform80, 2017-02-02:
Closing out old tickets; do feel free to re-open this if it's still an issue!

closed 2017-02-02 by waveform80