waveform80/picamera #231: Changing ISO doesn't appear to change sensitivity

dpcryer, 2015-07-12:
Firstly thanks for a wonderful library!

I'm probably missing something here but don't seem to be able to do much to control the sensitivity of the camera e.g. when I switch exposure_mode=off and then set the ISO to 100 then 400. I was expecting that this would change the analog_gain and digital_gain and therefore the exposure of the preview etc. assuming nothing else is changed. In my testing the preview exposure doesn't seem to change and neither do the values of the gains. I've included the status output from my program (just the values returned by calls to the picamera object)

exposure_mode set to off, iso set to 100

resolution:            2592,1944
framerate:             5
shutter_speed:         59984
exposure_speed:        59984
iso:                   100
exposure_mode:         off
exposure_compensation: 0
analog_gain:           83/16
digital_gain:          141/128

iso then set to 400, no change to analog_gain or digital_gain.

resolution:            2592,1944
framerate:             5
shutter_speed:         59984
exposure_speed:        59984
iso:                   400
exposure_mode:         off
exposure_compensation: 0
analog_gain:           83/16
digital_gain:          141/128

Am I missing something about what iso should be doing and how I should be adjusting the sensitivity.

Finally is there any way I can read/deduce the iso value when using auto-exposure (in the same way I can use exposure_speed to see what the automatically set value of shutter speed is) e.g. is iso=100*analog_gain*digital_gain?

lionelhadamar, 2016-05-03:
Hi, 
Thanks for this library, which is well documented, and easy to use for Raspberry Pi newbies as me.

I'd like to bump up this question, since I observe the exact same issue on my RPi 3B+ with camera Board v1.3.

As Dpcryer noted, it seems not possible to read the ISO value while the camera is in auto exposure mode. (the attribute getter returns the previously set ISO value, hence 0 if in auto exposure instead of the actual ISO value)

What I would like to do is set the camera on auto exposure mode, and then read the actual values of every attributes, and observe how the camera behaves under particular condtions.

Many thanks for considering this request

LionelCG7, 2016-05-04:
Hi,
First of all, thanks a lot for this well documented Python module, very handy for quick Pi camera setup.

I'd like to bump up Dpcryer's question regarding the access of ISO attribute while in auto exposure mode. 
As previously indicated, there is no way of knowing the ISO value while the exposure mode is set to 'auto' (the value is 0 in automatic mode, and the ISO getter yields 0 as well).
Would it be possible to read the camera parameters at any time, and no matter the camera mode ?

Moreover, what are the relation between the analog and digital gains and the ISO value ? In the documentation, it is written that when both gains equal 1, the ISO value is 100, but after some testing, I observe that analog_gain * digital_gain * 100 does not equals to the ISO value, and there are no correlation at all.

Thanks for considering this request,

waveform80, 2016-05-07:
The ISO value 0 simply means "auto". When you query the `iso` attribute you're simply querying the current ISO setting of the camera. If that's zero it simply means it's up to the camera to pick an appropriate ISO when it captures.

As to the correlation between ISO and the camera's gains - when camera's ISO is set to 100, the analog gain defaults to 1 (it may float higher though if the AGC algorithm, controlled by the `exposure_mode` property, determines that it needs to be higher). When camera's ISO is set to 200, the analog gain defaults to 2 (again, it can float around this as AGC adjusts things). What the gain limits are for each ISO, I don't know - you'd have to ask the firmware devs. Also, I don't expect the relationship is as simple as a linear multiplication given that the gains are both limited to a range of 1-8 overall.

waveform80, 2016-05-07:
Oh, the other point was: why do the gains not change when setting `iso` while `exposure_mode` is "off"? The answer is simply that `exposure_mode` "off" fixes the gains entirely - they won't budge at all (unless the camera is reset, e.g. by changing resolution / framerate). If you're looking for a way to control the gains manually there currently isn't one (not directly).

The best you can manage is to set `exposure_mode` to "auto" then fiddle with `iso` until the gains float to what you want (or close enough) and then set `exposure_mode` to "off" as fast as you can.

sbreuss, 2016-06-13:
Thanks for the explanation and the work around. It at least kind of works for me.

But are there any plan to implement a manual control for iso and the analog/digital gains?
The ability to set custom gains (or at least setting the iso value to control the gains) would be a really nice feature!

waveform80, 2016-06-13:
Currently, the firmware provides no means to set the gains manually (thus picamera, which is a relatively thin layer on top of MMAL cannot provide a means to do so either). I've not heard any plans for manual control of analog/digital gains - but that doesn't mean much one way or another. You're best off asking about that on the [camera forum](https://www.raspberrypi.org/forums/viewforum.php?f=43) where you're more likely to get a response from the firmware developers.

You can set the ISO value which *influences* the gains (as described above), though.

closed 2017-02-02 by waveform80