waveform80/picamera #234: JPEG thumbnails not working

wsat, 2015-08-25:
Hi,

I know this was an issue that was solved before but it seems it has popped up again. I am working with the python pi-timolo motion detection and capturing package:
https://github.com/pageauc/pi-timolo
To automatically create the thumbnails for a website, I am using the command (that's my only modification):

camera.capture(filename, format='jpeg', thumbnail=(200,100,35))

However, all pictures are captured with thumbnails being 64x48 pixels (below the output of EXIF.py):

Opening: motion/mo-cam1-1096.jpg
File has JPEG thumbnail
EXIF ApertureValue (Ratio): 6141/2000
EXIF BrightnessValue (Signed Ratio): 63/20
EXIF ColorSpace (Short): sRGB
EXIF ComponentsConfiguration (Undefined): YCbCr
EXIF DateTimeDigitized (ASCII): 2015:08:25 14:53:20
EXIF DateTimeOriginal (ASCII): 2015:08:25 14:53:20
EXIF ExifImageLength (Long): 1080
EXIF ExifImageWidth (Long): 1920
EXIF ExifVersion (Undefined): 0220
EXIF ExposureMode (Short): Auto Exposure
EXIF ExposureProgram (Short): Aperture Priority
EXIF ExposureTime (Ratio): 1349/1000000
EXIF FNumber (Ratio): 3623/1250
EXIF Flash (Short): Flash did not fire
EXIF FlashPixVersion (Undefined): 0100
EXIF FocalLength (Ratio): 4497/1250
EXIF ISOSpeedRatings (Short): 100
EXIF InteroperabilityOffset (Long): 876
EXIF MakerNote (Undefined): [101, 118, 61, 45, 49, 32, 109, 108, 117, 120, 61, 45, 49, 32, 101, 120, 112, 61, 49, 51, ... ]
EXIF MaxApertureValue (Ratio): 6141/2000
EXIF MeteringMode (Short): CenterWeightedAverage
EXIF ShutterSpeedValue (Signed Ratio): 4766947/500000
EXIF WhiteBalance (Short): Auto
Image DateTime (ASCII): 2015:08:25 14:53:20
Image ExifOffset (Long): 192
Image ImageLength (Long): 1080
Image ImageWidth (Long): 1920
Image Make (ASCII): RaspberryPi
Image Model (ASCII): RP_OV5647
Image ResolutionUnit (Short): Pixels/Inch
Image XResolution (Ratio): 72
Image YCbCrPositioning (Short): Centered
Image YResolution (Ratio): 72
Interoperability InteroperabilityIndex (ASCII): R98
Thumbnail Compression (Short): JPEG (old-style)
Thumbnail ImageLength (Long): 48
Thumbnail ImageWidth (Long): 64
Thumbnail JPEGInterchangeFormat (Long): 1012
Thumbnail JPEGInterchangeFormatLength (Long): 24576
Thumbnail ResolutionUnit (Short): Pixels/Inch
Thumbnail XResolution (Ratio): 72
Thumbnail YResolution (Ratio): 72

waveform80, 2016-05-07:
Can you let me know if this is still an issue? I've tried it with the latest firmware (881) and if seems happy to me:

```python
>>> from picamera import PiCamera
>>> camera.capture('foo.jpg', thumbnail=(400, 200, 35))
```

Then at the bash prompt (using a combination of tools from ImageMagick and ExifTool):

```bash
$ exiftool -b -ThumbnailImage foo.jpg > thumbnail.jpg
$ identify thumbnail.jpg
thumbnail.jpg JPEG 400x200 400x200+0+0 8-bit DirectClass 24.6KB 0.000u 0:00.000
```

One thing I'd check is exactly how you're reading the EXIF data; if you're using an old version of PIL, the EXIF support in there was thoroughly buggy (I vaguely recall playing with it in picroscopy and finding it just sent back the same values from the first file that was read) - I think later versions of Pillow (the now-supported fork) fixed that though.

waveform80, 2017-02-02:
Closing old issues; do re-open this if it's still a problem (though as noted above I suspect this is a problem with the way thumbnails are being read - I recently added some code to another ticket about extracting thumbnail data - I can dig that out if needed).

closed 2017-02-02 by waveform80