waveform80/picamera #235: preview_window(x,y,width,height) does not change add_overlay position

Joelbear5, 2015-08-26:
Default add_overlay() x,y location is centered on screen (when preview resolution is < screen resolution. When using preview_window(x,y,width,height) to update location (setting x,y to 0,0), the overlay is still centered default location on the screen while the preview window moved to upper left corner (0,0).

Solution is to update the overlay location using overlay location from preview_window location or have option to update overlay_window

waveform80, 2015-08-26:
Hmm, I'm not entirely clear what you're describing here - could you post a code snippet demonstrating what fails and what works?

Joelbear5, 2015-08-27:
The preview overlay layer image is not aligned with the camera preview_window. I have a 800x480 screen for reference. The preview overlay layer image is centered on screen while the preview_window is on the left side of the screen.

<pre><code>camera.resolution(640,480)
camera.preview_fullscreen = False
camera.preview_window(0,0,640,480)
camera.add_overlay(<i>'source image'</i>
                    layer=3,
                    (640,480),
                    alpha=128)
camera.start_preview
</code></pre>

The camera.preview_window preview stream is at `(0,0,640,480)` while the preview overlay image is centered on screen essentially at `(80,0,640,480)`.

I am expecting the overlay to follow the location of the `camera.preview_window()`, but it doesn't budge from the default location.

waveform80, 2015-08-29:
Ahh, now I see the issue, except that it's expected behaviour: overlays are not tied to the preview nor can they be. Overlays can exist without the preview, and vice versa.

Basically both previews and overlays are derivatives of the camera's "renderer" component. Renderers can be full screen or partial screen ("windowed" although they're nothing to do with windows on the desktop), and can have a static source (as in the case of an overlay) or a camera-based source (as in the case of the preview). Each renderer is independent of any other (which is quite useful in scenarios where overlays are used as interfaces).

This the reason that the `preview_fullscreen` and `preview_window` attributes are now deprecated: the `start_preview` method returns the preview renderer and you can set `fullscreen` and `window` on there, just as you can on the renderer returned by `add_overlay`. For example:

```python
import picamera

with picamera.PiCamera() as camera:
    camera.resolution = (640, 480)
    # Set up the preview. Here we're using the return value of start_preview
    # but you can specify these values as arguments to start_preview too
    preview = camera.start_preview()
    preview.fullscreen = False
    preview.window = (0, 0, 640, 480)
    preview.alpha = 255
    # Set up the overlay. Again, you can specify arguments to add_overlay
    # or set attributes on the resulting renderer
    overlay = camera.add_overlay(source_img.tostring(), size=(640, 480))
    overlay.fullscreen = False
    overlay.window = (0, 0, 640, 480)
    overlay.alpha = 128
    overlay.layer = 3
```

Hope that clears things up!

Joelbear5, 2015-08-29:
Thanks! `overlay.window()` was not in the documentation so I didn't know it existed, but this is perfect. Thanks!

waveform80, 2015-08-29:
Actually it is in the docs, although I'll admit it is a bit obscure. The [`add_overlay`](http://picamera.readthedocs.org/en/release-1.10/api_camera.html#picamera.camera.PiCamera.add_overlay) method states:

> ... each call to `add_overlay()` returns a new `PiOverlayRenderer` instance representing the overlay ...

The [`PiOverlayRenderer`](http://picamera.readthedocs.org/en/release-1.10/api_renderers.html#picamera.renderers.PiOverlayRenderer) docs then state:

> ... This class descends from `PiRenderer` and adds a static source for the MMAL renderer.

And finally, [`PiRenderer`](http://picamera.readthedocs.org/en/release-1.10/api_renderers.html#picamera.renderers.PiRenderer) documents all the attributes common to overlay and preview renderers (as it's the parent class for each). I could expand [the recipe](http://picamera.readthedocs.org/en/release-1.10/recipes1.html#overlaying-images-on-the-preview) a bit, but the recipes are really only there to introduce a topic and show-case some common techniques - documenting every single method and property is the job of the reference section (which the recipes link to fairly liberally).

closed 2016-05-07 by waveform80