waveform80/picamera #238: Applying a filter to the Pi Camera Image

TeCoEd, 2015-09-09:
Hi, I wondered if there was a method for taking a picture and applying a filter such as sepia, negative within the Pi Camera code, or would this require additional software?
Many thanks,
Dan

waveform80, 2015-09-09:
You can use [image_effect](http://picamera.readthedocs.org/en/release-1.10/api_camera.html#picamera.camera.PiCamera.image_effect) to apply a filter prior to taking the picture (it'll affect the preview, and the capture itself). The filters are defined by the firmware (so the filters are non-editable) but there are [parameters](http://picamera.readthedocs.org/en/release-1.10/api_camera.html#picamera.camera.PiCamera.image_effect_params) for controlling various filter options.

TeCoEd, 2015-09-13:
Many thanks Dave, you helped me to get it working
Regards

closed 2016-05-07 by waveform80