waveform80/picamera #239: Sub-macroblocks of 8x8 and 4x4

edl26, 2015-09-20:
Hi,

I'm trying to work with sub-macroblocks of 8x8 and 4x4 apart from the 16x16 standard macroblock size. This is, in some way, to be more accurate about the areas in which movement is detected. I've seen that it is possible to change the intra search mode of the encoder through MMAL_VIDEO_ENCODE_H264_MB_INTRA_MODES_T but I'm not sure about how I should do it.

Many thanks

6by9, 2015-09-28:
Really the wrong place to be asking - either https://github.com/raspberrypi/userland or the forums on raspberrypi.org would have been better. 

MMAL parameter ```MMAL_PARAMETER_VIDEO_ENCODE_H264_MB_INTRA_MODE``` takes a ```MMAL_PARAMETER_VIDEO_ENCODER_H264_MB_INTRA_MODES_T``` structure, of which the ```mb_mode``` member is an ```MMAL_VIDEO_ENCODE_H264_MB_INTRA_MODES_T```. 
It is treated as a bitmask (hence the values assigned to the enum values) which appear to control which macroblock modes are permitted. 0 is treated the same as 7 (all modes allowed), and that is the default.

edl26, 2015-10-04:
Thanks for the answer, I asked here because my idea is to work with this library and python. I'll take a look in the other two places.

closed 2017-02-02 by waveform80