waveform80/picamera #243: Streaming IOError: cannot identify image file

xmen4u, 2015-10-13:
Tried the example that worked earlier... something must have updated. Now i get:
IOError: cannot identify image file<_io.BytesIO object at 0x1058b40b0>

code:

server:
import io
import socket
import struct
from PIL import Image

# Start a socket listening for connections on 0.0.0.0:8000 (0.0.0.0 means
# all interfaces)
server_socket = socket.socket()
server_socket.bind(('0.0.0.0', 8000))
server_socket.listen(0)

# Accept a single connection and make a file-like object out of it
connection = server_socket.accept()[0].makefile('rb')
try:
    while True:
        # Read the length of the image as a 32-bit unsigned int. If the
        # length is zero, quit the loop
        image_len = struct.unpack('<L', connection.read(struct.calcsize('<L')))[0]
        if not image_len:
            break
        # Construct a stream to hold the image data and read the image
        # data from the connection
        image_stream = io.BytesIO()
        image_stream.write(connection.read(image_len))
        # Rewind the stream, open it as an image with PIL and do some
        # processing on it
        image_stream.seek(0)
        image = Image.open(image_stream)
        print('Image is %dx%d' % image.size)
        image.verify()
        print('Image is verified')
finally:
    connection.close()
    server_socket.close()

client:
import io
import socket
import struct
import time
import picamera

# Connect a client socket to my_server:8000 (change my_server to the
# hostname of your server)
client_socket = socket.socket()
client_socket.connect(('my_server', 8000))

# Make a file-like object out of the connection
connection = client_socket.makefile('wb')
try:
    with picamera.PiCamera() as camera:
        camera.resolution = (640, 480)
        # Start a preview and let the camera warm up for 2 seconds
        camera.start_preview()
        time.sleep(2)

        # Note the start time and construct a stream to hold image data
        # temporarily (we could write it directly to connection but in this
        # case we want to find out the size of each capture first to keep
        # our protocol simple)
        start = time.time()
        stream = io.BytesIO()
        for foo in camera.capture_continuous(stream, 'jpeg'):
            # Write the length of the capture to the stream and flush to
            # ensure it actually gets sent
            connection.write(struct.pack('<L', stream.tell()))
            connection.flush()
            # Rewind the stream and send the image data over the wire
            stream.seek(0)
            connection.write(stream.read())
            # If we've been capturing for more than 30 seconds, quit
            if time.time() - start > 30:
                break
            # Reset the stream for the next capture
            stream.seek(0)
            stream.truncate()
    # Write a length of zero to the stream to signal we're done
    connection.write(struct.pack('<L', 0))
finally:
    connection.close()
    client_socket.close()

waveform80, 2015-10-13:
Do you receive that error immediately without any images being decoded at all? If so, I'd suspect something wrong with the sending of the image length, or the checking of it because that error is usual when the stream being read from is blank.

xmen4u, 2015-10-13:
@waveform80 yea, no images being decoded at all. Well this is the exact code from:
http://picamera.readthedocs.org/en/latest/recipes1.html

Which worked before. Maybe some big endian change ? idk.. something broke

waveform80, 2015-10-14:
Can't be an endian issue - in the both the scripts you've pasted the format is "<L" which is a little-endian unsigned long (4-bytes). Are those scripts just copied from the docs or are they the actual scripts you're running?

xmen4u, 2015-10-15:
Copied + tried running. Same script worked before. Something changed probably after apt-get update . Try them yourself

waveform80, 2015-10-15:
I have, they work perfectly for me, after changing the `'my_server'` hostname obviously. I'll try an `apt-get update` + `apt-get upgrade` to see if anything breaks, but it never has before so I'm rather dubious that's the cause.

waveform80, 2015-10-15:
Okay, updated two pis, one running Raspbian Wheezy, the other running the new Raspbian Jessie. Both work fine with the scripts you've posted. Are those the script's you're *really* running or were they just copied and pasted straight from the docs for convenience? (I'm assuming so given `'my_server'` hasn't been replaced in the client script).

xmen4u, 2015-10-15:
I've ran the code as-is from the docs - of course i've changed the client 'my_server' to my ip. 
output of client code:
```
File "test.py", line 28, in <module>
    image = Image.open(image_stream)
  File "/Library/Python/2.7/site-packages/PIL/Image.py", line 2286, in open
    % (filename if filename else fp))
IOError: cannot identify image file <_io.BytesIO object at 0x10bbac0b0>
```
output of server:
```
Traceback (most recent call last):
  File "app4.py", line 44, in <module>
    connection.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
socket.error: [Errno 32] Broken pipe
```

client: 
```
uname -a 
```
output:
```
Linux raspberrypi 3.18.13-v7+ #784 SMP PREEMPT Sat May 9 15:57:36 BST 2015 armv7l GNU/Linux
```

also:
```
"python-picamera is already the newest version."
```

I don't know why... :/ 
how can i get to the bottom of this? not saying there is a bug in the code... but something isn't working for me. i've also tried uninstalling + installing both on the client and on the server. with out a change

waveform80, 2015-10-15:
Hmm ... this is a weird one. There's a few possibilities I can think of off the top of my head. The first is simple: the network connection is so bad it can't even write a single file properly. But that's rather doubtful (if it's that bad, how would it even make the connection?).

The second is that there's some issue with your Pi's firmware. While picamera's up to date, you've got a slightly old firmware (`#784` whereas I think `#817` is the latest at the moment). You can try running `sudo rpi-update` to upgrade the firmware and then reboot afterward. I'm also dubious about that being the reason though ... that would imply the firmware you're on is so broken it can't even write a JPEG properly (and I've never seen that!). For reference, both Pi's I've tested on have firmware `#817` installed.

The third possibility is that your Mac has a broken PIL installation which is having trouble interpreting JPEGs from a BytesIO stream. To test this one, you can try running the following code on your Mac (change "some_image.jpg" to point to any old JPEG you've got lying around):

```python
import io
from PIL import Image

image_file = io.open('some_image.jpg', 'rb')
image_stream = io.BytesIO()
image_stream.write(image_file.read())
image_stream.seek(0)
Image.open(image_stream)
```

If that blows up with an error, the version of PIL you've got installed is broken / incredibly ancient (I'm afraid I can't help much with that - I'm not a Mac expert, but I'm told "homebrew" is considered the best way of getting things like Python and libraries installed these days).

xmen4u, 2015-10-16:
@waveform80 So! good news! 
1) 1st suggestion - couldn't be as they are connected via network cable and also connection speed is great (Verified).
2) i've tried the 3rd suggestion - which worked and didn't blow up :)
2) then i moved to rpi-update

reboot , ran , WORKED!
didn't change a thing. 
I can tell you that the code worked like a month ago. Probably "sudo apt-get update + upgrade" borke something along the way that a firmware update was needed for core video (which was part of the firmware upgrade). 

Thank you SOOO MUCH. It's good to know this was the cause.

waveform80, 2015-10-17:
Ah, glad we got that sorted. Quick follow-up question - have you ever used rpi-update before? The reason I ask is that basically there's two ways of updating firmware: the usual apt-get route and the rpi-update route, and given the issue was obviously with the firmware I'm intrigued to know if it was likely to be one obtained from apt-get or rpi-update.

The apt-get route is updated in a pretty conservative fashion so it tends to be a good way behind the "bleeding edge". However, it's generally good enough for most and it's definitely the "proper" way in as much as it means the system package manager is aware of all the firmware files.

The rpi-update route is updated much more frequently and, historically, was extremely useful for keeping up with the all the latest camera functionality (that's not so much the case these days, but it's still "ahead" of the apt-get firmware). However, it's somewhat frowned upon because it stomps over the files that apt-get thinks its maintaining. That generally doesn't matter but it can cause issues in some rare cases.

So, if it was a firmware from rpi-update I wouldn't be too surprised at issues in it (it's happened, rarely, in the past). However, if you've never used rpi-update that'd mean it was a firmware from apt-get. Firstly, that'd be rather surprising (I get the impression rather more testing goes on with stuff that goes into apt-get), but secondly it'd mean it'll affect a lot more people and I'll need to keep an eye out for firmware `#784` in future!

closed 2016-05-07 by waveform80