waveform80/picamera #244: equivalent of --stats raspistill option

ogre, 2015-10-14:
Hello
From raspistill man:
-st, --stats    : Force recomputation of statistics on stills capture pass

What is the equivalent of this in picamera module ?

thanks
-michal

waveform80, 2015-10-14:
That's the [`still_stats`](http://picamera.readthedocs.org/en/release-1.10/api_camera.html#picamera.camera.PiCamera.still_stats) parameter of the PiCamera class.

waveform80, 2015-10-14:
Sorry, still_stats attribute I meant to say (set it after construction).

ogre, 2015-10-14:
Thank you. I was sure I checked the docs, apparently not :/

waveform80, 2015-10-14:
No worries - I'd suggest giving the search feature a try with individual words. It's not great with phrases (unfortunately that's more than one can realistically ask of a browser based engine at the moment), but with straight forward one-word queries it's generally fairly good. For example, if you know the attribute you want up front "still_stats" will provide it as the first hit. Likewise, so will "stats" (again, because it appears in the title).

Unfortunately more complex phrases like "statistics" fail to find the entry directly (it directs to the parent topic, but given that's the PiCamera API reference, I'd never blame anyone for not finding "still_stats" from that huge result). Basically the search function isn't fantastic, but if you treat it like a really dumb version of google (only able to handle individual words) it's not too bad.

closed 2016-05-07 by waveform80