waveform80/picamera #245: Is it possible to extend a PiCameraCircularIO dynamically (using truncate())?

maximedupre, 2015-10-18:
In the documentation for `truncate`, it says that is can:
> Resize the stream to the given size in bytes (or the current position if size is not specified). This resizing can extend or reduce the current stream size.

However, when I try, I can only reduce the current stream size. If I try to extend it, the size stays the same:

    DEFAULT_BYTES_PER_SECOND = 2125000
    stream = picamera.PiCameraCircularIO(camera, size=DEFAULT_BYTES_PER_SECOND * 20) #42500000
    stream.truncate(size=stream.size + DEFAULT_BYTES_PER_SECOND)
    stream.size # 42500000

Is there something I am doing wrong?

Thank you!

maximedupre, 2015-11-11:
It isn't possible to extend a stream further than it's initial/maximum size (passed at instantiation).

waveform80, 2015-11-11:
Sorry, I've been rather too busy to look at this over the last month. I'll try and have a go at it this week, but it does sound like a bug (at the very least `truncate` ought to match its documentation in terms of return value).

waveform80, 2016-03-18:
Ahh, I should've noticed this the first time I looked at it. I'm afraid this isn't a bug: you just can't extend the underlying storage of the circular stream using `truncate` beyond the limit set in the constructor (the maximum size of the stream, which is what the `size` property reports, is fixed and unchangeable).

You can use `truncate` as you would in a normal stream to extend the content of the stream beyond its nominal end, but the entire point of the circular stream is that once its storage exceeds some boundary (given by `size`, which isn't a "normal" stream attribute) it deliberately "loses" data by overwriting its start. To demonstrate this, we can construct a circular stream with a size-limit of 10 bytes, use truncate to extend the empty stream to 5 bytes long, then write 5 bytes of data filling the rest of the storage, then write 5 more which will cause the 5 NULL bytes at the beginning to be "lost", then finally use truncate to extend the stream with 5 more NULL bytes:

```python
>>> import picamera
>>> stream = picamera.CircularIO(size=10)
>>> stream.getvalue()
''
>>> stream.seek(5)
5
>>> stream.truncate()
>>> stream.getvalue()
b'\x00\x00\x00\x00\x00'
>>> stream.write(b'abcde')
5
>>> stream.getvalue()
b'\x00\x00\x00\x00\x00abcde'
>>> stream.write(b'fghij')
5
>>> stream.getvalue()
b'abcdefghij'
>>> stream.tell()
10
>>> stream.seek(15)
15
>>> stream.truncate()
>>> stream.getvalue()
b'fghij\x00\x00\x00\x00\x00'
```

Note that the final truncate does "extend" the stream in the sense that it's equivalent to writing 5 NULL bytes to the end of the stream (which is the normal meaning of truncate when it's used with the file pointer beyond the end of the file) but that in the case of the circular stream it doesn't actually extend the underlying storage which is limited to the value given to the constructor.

closed 2016-03-18 by waveform80