waveform80/picamera #246: Recording Audio with Video

JpEncausse, 2015-10-18:
Any tips, advice, on how to record audio from a microphone with the video ?

waveform80, 2016-05-07:
Can't say I've ever tried muxing audio into picamera's video output - but it's certainly something various people have asked about. I'll leave this open as a question in case anyone has any good recipes or tips - it'd be worth adding something to the doc's recipes if so.

lwilli, 2019-08-26:
Is there any update on supporting audio? I saw mentions of audio in https://github.com/waveform80/picamera/pull/490, but I'm not sure if/how it addresses this. 

I have an i2s microphone that can successfully record audio, but I haven't been able to figure out how to combine it with picamera's video. Maybe it's possible by using https://picamera.readthedocs.io/en/release-1.13/api_mmalobj.html?highlight=audio#picamera.mmalobj.MMALAudioPort somehow? If someone can shed some light on how we might be able to do this, it would very helpful! Thanks.

dinapappor, 2020-09-16:
I too am interested in this. @dbbnrl Could you shed some light on how to use this? :pray: