waveform80/picamera #248: Invalid PCR value in ES_OUT_SET_(GROUP_)PCR

Redsandro, 2015-11-14:
```
connection = server_socket.accept()[0].makefile('wb')
camera.start_recording(connection, format='h264', profile='high', quality=0, bitrate=20000000))
camera.wait_recording(60)
```

Using

```
vlc tcp/h264://192.168.1.5:5000/
```

Gives stutters and these `vlc` errors:

```
[00007f2df80009b8] core input error: Invalid PCR value in ES_OUT_SET_(GROUP_)PCR !
[00007f2df80009b8] core input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
[00007f2df80009b8] core input error: ES_OUT_RESET_PCR called
[00007f2df80009b8] core input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1037 ms)
[00007f2df80009b8] core input error: ES_OUT_RESET_PCR called
[00007f2df80009b8] core input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1040 ms)
[00007f2df80009b8] core input error: ES_OUT_RESET_PCR called
[00007f2df80009b8] core input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1051 ms)
[00007f2df80009b8] core input error: ES_OUT_RESET_PCR called
```

etc.

What causes this? Can I fix this?

Also, without `wait_recording` it seems to happen a lot more often. Any idea why?

waveform80, 2016-05-07:
It may be that as the H.264 provided by the camera's video encoder doesn't contain frame-rate info (and I'm not sure if it has valid PTS) that vlc's trying to play stuff "too fast". You could try adjust the camera's framerate down to vlc's default (24 or 25 - I forget which), but I'm not sure how much it'll help (those console errors are pretty typical with VLC when playing back recordings straight from picamera without encapsulating them with MP4Box first).

waveform80, 2017-02-02:
Closing old issues; do feel free to re-open if you've any further queries regarding this!

closed 2017-02-02 by waveform80