waveform80/picamera #250: Event driven socket integration

wardjk, 2015-11-19:
I need to occasionally connect to then disconnect from an unattended Pi to remotely get a video stream. The example in “Recording to a network stream” with Pi acting as the server is similar to my needs except it doesn’t allow clients to repeatedly connect. My project uses PyQt4 so I first used PyQt4's support for sockets but they are created on a different thread which causes an error when the file-like object is passed to start_recording(). I then tried a socket using asyncore but found that once asyncore starts polling, it can’t be stopped and effectively blocks. I’ve never used Twisted and don’t know if it would solve my problem. Do you know of a solution for this dilemma?

guyver2, 2015-11-24:
On the server side you just have to insert the code from the example inside a higher level infinite loop that will :
1) accept a new connection
2) stream images
3) close the connection when the client goes away

```python
import io
import socket
import struct
from PIL import Image

# Start a socket listening for connections on 0.0.0.0:8000 (0.0.0.0 means
# all interfaces)
server_socket = socket.socket()
server_socket.bind(('0.0.0.0', 8000))
server_socket.listen(0)

# unstoppable loop
while True : 
    # Accept a new connection and make a file-like object out of it
    connection = server_socket.accept()[0].makefile('rb')
    try:
        # loop that sends images on the socket
        while True: 
            # Read the length of the image as a 32-bit unsigned int. If the
            # length is zero, quit the loop
            image_len = struct.unpack('<L', connection.read(struct.calcsize('<L')))[0]
            if not image_len:
                break
            # Construct a stream to hold the image data and read the image
            # data from the connection
            image_stream = io.BytesIO()
            image_stream.write(connection.read(image_len))
            # Rewind the stream, open it as an image with PIL and do some
            # processing on it
            image_stream.seek(0)
            image = Image.open(image_stream)
            print('Image is %dx%d' % image.size)
            image.verify()
            print('Image is verified')
    finally:
        connection.close()

# note how this line goes out of the loop as we still want the server to listen for new connections
server_socket.close() 
```

As for the client side, you can use the regular code given in the example. The server needs to be launch only once and you can connect as many client as you want (sequentially).
I had that piece of code running on my Pi for weeks.

guyver2, 2015-11-24:
You'll need to invert the server/client role distribution given in the example. Make the server run on the Pi and add an extra infinite loop so that once a connection is lost, instead of closing the server_socket, it starts listening again for new connections.

wardjk, 2016-02-27:
There are two examples in “Recording to a network stream” and I am using the one with Pi acting as the server but it has the drawbacks I mentioned. Running an infinite loop results in blocking which is the problem I’m trying to avoid.

guyver2, 2016-02-28:
hence the **extra** loop around the first infinite one.
the example gives something like this 
```
wait_for_connection
accept
while(True):
    send_image
```

and you need to write something like this, You cannot just use the given example as is. 

```
while(True): # first infinite loop to reconnect
    wait_for_connection
    accept
    while(True and connection_alive): # second infinite loop to send image
        send_image
```

wardjk, 2016-02-29:
The challenge isn't only with reconnecting. Loops like this result in code blocking. An extra loop doesn't negate this which is why I am still looking for a non-blocking solution.

guyver2, 2016-02-29:
Well I must admit I don't see where that would be blocking... If you exit the inner loop (and shut down the camera) as soon as the current connection is closed. Your program will come back to a state of waiting for a new connection. Ready to stream again whenever something comes and ask for it.

I'm using this achitecture in one of my project and can easily let the pi run for weeks and connect to it twice a day for a few minutes (sometimes hours)

wardjk, 2016-02-29:
A distinguishing advantage of PiCamera is the ability to change camera properties live. I'm not aware of a practical way to change properties while a loop is running.

guyver2, 2016-02-29:
Well for that I use another socket in another thread which holds a pointer to the current picamera object. 
I have one socket for the stream, and another one for client/server communication, mainly for live camera setup (some of the parameters require a reboot of the camera though)

wardjk, 2016-04-22:
I submitted a related issue titled [Change camera attributes while streaming](https://github.com/waveform80/picamera/issues/276) which is a little broader than this one.

wardjk, 2016-04-26:
Problem solved and explained in [Change camera attributes while streaming](https://github.com/waveform80/picamera/issues/276). No loops used.

closed 2016-04-26 by wardjk