waveform80/picamera #251: What could cause this stack trace?

alexkreimer, 2015-11-26:
Traceback (most recent call last):  
File "/home/pi/rpi-repo/rpi/worker.py", line 122, in detect_motion 
   if camera.Camera.instance().detect_motion():
File "/home/pi/rpi-repo/rpi/camera.py", line 158, in detect_motion
    camera.start_recording('/dev/null', format='h264', motion_output=output)
File "/home/pi/.virtualenvs/env/local/lib/python2.7/site-packages/picamera/camera.py", line 1245, in start_recording
    encoder.start(output, options.get('motion_output'))
File "/home/pi/.virtualenvs/env/local/lib/python2.7/site-packages/picamera/encoders.py", line 1235, in start
    super(PiVideoEncoder, self).start(output)
File "/home/pi/.virtualenvs/env/local/lib/python2.7/site-packages/picamera/encoders.py", line 826, in start
    prefix="Failed to enable encoder output port")#012  File "/home/pi/.virtualenvs/env/local/lib/python2.7/site-packages/picamera/exc.py", line 191, in mmal_check
    raise PiCameraMMALError(status, prefix)#012PiCameraMMALError: Failed to enable encoder output port: Out of memory

waveform80, 2015-11-26:
Quite a few things including trying to encode H264 at a resolution > 1080p or if your GPU mem-split is too small for whatever you're trying to do?

alexkreimer, 2015-11-26:
```python
    def detect_motion(self):
        with picamera.PiCamera() as camera:
            camera.resolution = (640, 480)
            camera.framerate = Fraction(15,1)
            with MotionData(camera, self.data) as output:
                #camera.start_recording('video_%d.h264' % self.count, format='h264', motion_output=output)
                camera.start_recording('/dev/null', format='h264', motion_output=output)
                camera.wait_recording(2)
                camera.stop_recording()
                self.data = output.data
```

waveform80, 2017-02-02:
Sorry, looks like this one slipped through the cracks! I'm just cleaning out old issues but it looks like I didn't investigate this one as I couldn't run the script provided (too many missing pieces), and I forgot to post a message requesting more detail. Given the ticket's over a year old I'll close it for now but do feel free to re-open it (e.g. if you can provide a standalone script I can use to re-produce the issue)

closed 2017-02-02 by waveform80