waveform80/picamera #252: Changing position of annotation text

myrtleTree33, 2015-12-10:
Hi, 

Is there any way to set the XY position of annotations?

Great job by the way!!

waveform80, 2015-12-10:
Sorry, no. The firmware currently provides no means of positioning annotations (just text size and colour).

closed 2016-05-12 by waveform80

barsznica, 2017-08-05:
Setting x,y positions would be great!
Thanks for the great work;)

waveform80, 2017-08-05:
Sorry, not up to me - annotation is a facility provided by the firmware (picamera is just a wrapper over (some libraries which wrap) the firmware).

barsznica, 2017-08-20:
Do you know how I could use colours other than white for something like below?

# Create an array representing an image of a cross through the center of the 
# display. The shape of the array must be of the form (height, width, color)
a = np.zeros((720, 1280, 3), dtype=np.uint8)
a[640, :, :] = 0xff
a[:, 360, :] = 0xff