waveform80/picamera #256: Raspistill-like timeout option?

ericmjl, 2016-01-15:
Before I go on, just want to say, great work on the Picamera module! It's made rapid prototyping much easier for me, and I'm using it as one part of my thesis work. :smile:

The problem I'm encountering is over-exposed images outdoors. I think it would be in the style of what someone posted in this forum: http://www.thefruitycomputer.com/forums/topic/789-overexposure-problem-with-pi-camera-module/

Is there a way to set a `timeout` option, or might I be missing/overlooking something in the Picamera documentation?

Once again, big kudos for the package! :smile:

andreroggeri, 2016-02-18:
Maybe you could look at the [exposure docs](https://picamera.readthedocs.org/en/release-1.10/recipes1.html?highlight=exposure)

Although I am not very expert

ericmjl, 2016-02-18:
Just figured it out - manually use Python's sleep function to sleep for a bit first.

closed 2016-02-18 by ericmjl