waveform80/picamera #258: camera = picamera.PiCamera issue

ameyring, 2016-02-01:
When I run the command "camera = picamera.PiCamera()" in either Python, Python3, or as part of a program, I keep getting this message:

Traceback (most recent call last):
  File "<pyshell#3>", line 1, in <module>
    camera = picamera.PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 489, in __init__
    self._init_defaults()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 647, in _init_defaults
    prefix="Failed to get annotation test")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 191, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to get annotation test: Function not implemented

"python3" is "python" when running that version. I have the latest updates and everything as far as I can tell. Camera is also enabled. Can't get past this error. Hope you can help. Thanks.

andreroggeri, 2016-02-18:
Is raspivid/raspistill working ? Also, are you using raspbian ?

If you could post the output of `uname -a` (You can take a look at [Firmware Upgrade Section](https://picamera.readthedocs.org/en/release-1.10/install3.html#firmware-upgrades)

waveform80, 2016-05-07:
This sounds like you're running a very old firmware - as André mentioned, upgrading firmware ought to fix it.

waveform80, 2017-02-02:
Closing old issues; feel free to re-open if you have any further queries about this!

closed 2017-02-02 by waveform80