waveform80/picamera #259: Horizontal line appearing in video at certain resolutions when annotate_frame_num=True

bennettrogers, 2016-02-08:
When capturing h264 video with annotate_frame_num set to True, a horizontal line will appear in the video at certain resolutions. For example, the line shows up when capturing at 1296x972 but not at 1296x730. Is there a way to prevent this line artifact from appearing?

This is an issue for me since I'm trying to stream video and capture images simultaneously, and the artifact appears when the camera resolution is set to the full 2592x1944 resolution, even if a resize is applied to the video capture.

I'm attaching a still from a video where the line can be seen.

Thank you!

[image]

6by9, 2016-02-11:
The line will always be there, independent of resolution.
Rendering graphs for exposure time, analogue gain, lens position, and adding frame_num are all lumped together in one function. At the end the firmware draws a grey line at the bottom to show where the end of the graph is - that's the line you're seeing.

Not a Picamera issue. Raise one under https://github.com/raspberrypi/firmware/issues if you want a fix, but it won't be a priority.

bennettrogers, 2016-02-11:
Ok thank you, it's good to know this is in the camera firmware and not in picamera, and that it's intentional.

The visibility of the line definitely does depend on the resolution though. When frame number annotations are on, the line is obvious at 1296x972 and invisible at 1296x730. Maybe it comes down to an aliasing effect or something, causing the line to appear thicker or thinner (or invisible) depending on the aspect ratio.

6by9, 2016-02-11:
Most likely aliasing, or there's a bug in the fill routine (it fills a rectangle which is two lines high to try to avoid fun with YUV chroma subsampling).
File a bug report on the firmware and it'll get added to my list of things. There's no reason why having the frame number on requires the graph bottom line (it makes sense for the graphing options only), and it's the sort of 10minute fix I tend to go looking for when I have just a few spare minutes.

bennettrogers, 2016-02-22:
The latest rpi firmware resolves this issue.

closed 2016-02-22 by bennettrogers