waveform80/picamera #260: Add PiYArray?

josharian, 2016-02-11:
First, thanks for this incredible library and for such thorough, readable docs.

Many computer vision applications use only greyscale input. Extracting the Y plane out of PiYUVArray is wasteful, though, since it's a fair amount of work to construct the YUV array in the first place.

In just a few minutes, I cobbled together a PiYArray implementation that mimics PiYUVArray but never constructs the color planes. I'm not doing much other work (yet), but it almost doubled my framerate.

PiYArray is easy to build, but I think it might be of broad enough to use to warrant being included. If you'd like to see it included, I'd be happy to send a pull request.

kif, 2016-02-27:
Sounds like a good idea. Can you provide the patch via a pull request ?

laser101, 2016-03-01:
This would be very useful as applications often approximate image intensity using the Y component.

kif, 2016-03-01:
On Tue, 01 Mar 2016 05:28:52 -0800
Russ <notifications@github.com> wrote:

> This would be very useful as applications often approximate image intensity using the Y component.

I wonder if it would not be better to have a unic container which would have:
* rgv_array
* yuv_array
* y_array
* raw_array (?)
properties, all of them lazy evaluated and memoized as rgb_array for the YUV-container.

josharian, 2016-03-02:
I'd rather keep the current separate class structure, but API decisions are better kept in one hand, so I'll defer to your preferences. Let me know what they are, and I'll pull together a PR, although it may take a week or three.

josharian, 2016-03-04:
In case you decide you want to move forward without me on this (if I am too slow), here's an out-of-tree, doc-less code snippet:

```python
class PiYArray(picamera.array.PiArrayOutput):

	def flush(self):
		super(PiYArray, self).flush()
		resolution = self.size or self.camera.resolution
		fwidth, fheight = picamera.array.raw_resolution(resolution)
		y_len = fwidth * fheight
		uv_len = (fwidth // 2) * (fheight // 2)
		data = self.getvalue()
		if len(data) != (y_len + 2 * uv_len):
			width, height = resolution
			raise PiCameraValueError('Incorrect buffer length for resolution %dx%d' % (width, height))
		self.array = np.frombuffer(data[:y_len], dtype=np.uint8)
		self.array.shape = (fheight, fwidth)
```

waveform80, 2016-03-18:
Before anybody puts any significant time into this, I'd have a quick look at what's coming up in 1.11 (which I've finally made some progress on - if I can close off #218 next week I'm planning to go for a release shortly after). Specifically, have a look at the "[capture to buffer](http://picamera.readthedocs.org/en/latest/recipes2.html#capturing-to-a-numpy-array)" functionality (#241) and how it can be used to capture just the Y plane with minimal overhead (see the end of [the YUV recipe](http://picamera.readthedocs.org/en/latest/recipes2.html#unencoded-image-capture-yuv-format)).

I suspect the new capture-to-buffer is going to make the various Pi*Stream classes fairly redundant (though the analysis classes covering unencoded video will still be useful). Obviously I'll leave 'em around for backward compatibility but I don't see much use for them when you can capture straight into a numpy array.

waveform80, 2016-05-07:
Argh - sorry 1.11's taking so long - nearly there though, and in more good news for people wanting to capture for analysis the new code is a lot faster at doing RGB/BGR captures under certain circumstances too (depends on what pipeline's being constructed though).

waveform80, 2017-02-02:
1.11 implemented the capture to buffer functionality as detailed above (including the ability to capture "just the Y values"). If people still want a PiYArray, do feel free to re-open this (or post a pull-request); I'm happy to add something like that for the sake of completeness, but I think capture to buffer is probably quicker and easier.

closed 2017-02-02 by waveform80