waveform80/picamera #265: Camera LED control unlikely to work on Pi3

6by9, 2016-03-02:
Just a heads up.
The Camera LED GPIO has moved again on Pi3, so the current logic is unlikely to do the right thing. I haven't worked out the detail of controlling it as yet, or whether it is even possible on Pi3.

I'll update if/when I find anything.

6by9, 2016-03-02:
Confirmed that the camera LED line has moved to a GPIO expander that is not directly accessible from the ARM.
I want to get to the shutdown line too, so will be looking at extending https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/gpio/gpio-bcm-virt.c to support the camera GPIOs too.

waveform80, 2016-03-02:
Ahh, thanks for the heads up! After a wild couple of days of picademy, and several months neglecting picamera, I'm going to try and plough through as much of the outstanding tickets as I can tomorrow (really want to try and get a release out before the next one as there's some bugs I don't want to see come up yet again :)

AndySchroder, 2016-05-15:
I can confirm that the LED control is not working on raspberry pi 3 with the v1 camera. The red LED just stays on all the time.

6by9, 2016-06-07:
Pi3 LED control is being discussed on https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137&start=100#p989832
If someone has the energy, then it ought to be wrapped into a full GPIO driver, and you should then control it via /sys/class/gpio/gpioXXX/value. Otherwise you'll have to drive the mailbox directly yourself.

waveform80, 2016-06-13:
For now, I'll just stick a note in the docs about LED control on the Pi3 (this is in the "more effort than it's worth to fix" category :). If a GPIO driver comes along at some point, I'll add in a fix for it.

6by9, 2016-06-13:
I can relate to that category.

FYI Code posted at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137&start=100#p990152 based on Akane's sample app at https://github.com/Terminus-IMRC/rpi3-gpiovirtbuf
The RPI_FIRMWARE_SET_GPIO_STATE call will work with any GPIO, not just the expander, just use the raw GPIO number without an offset of 128.
It would be worth double checking the permissions on the mailbox service (/dev/vcio), as it may be possible to use it without root. 

A full GPIO driver is likely to be a way down the list of tasks, although it's not a task that needs to be done by someone with firmware access until it gets into adding direction control.

Terminus-IMRC, 2016-06-16:
FYI: Eric Anholt (@anholt) is planing to write full GPIO driver using the Mailbox calls: http://permalink.gmane.org/gmane.linux.kernel.rpi/3943

6by9, 2016-06-16:
Thanks for the heads up. (I've now subscribed to that mailing list)

waveform80, 2017-02-02:
Ooops - completely forgot to add the note to the docs. I'll add that in a mo, and leave this open just in case a GPIO driver appears (although with the V2 module lacking an LED I suspect the use-case for this to diminish over time).

AFUDIrk, 2017-02-17:
Hello,
Has two Pi 2 b and one Pi 3.
In the two Pi 2 I can switch the LED from the CSI camera module via GPIO 32 on and off.
Turn on an IR cut filter.
echo 32 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio32/direction
echo 1 > /sys/class/gpio/gpio32/value
echo 0 > /sys/class/gpio/gpio32/value
How can I switch the LED (IR cut filter) at Pi 3?

waveform80, 2017-02-17:
@AFUDIrk I'd suggest reading the forum post 6by9 linked to above: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137&start=100#p989832

6by9, 2017-02-28:
There is a kernel driver waiting to be merged at https://github.com/raspberrypi/linux/pull/1852 that should allow you to control the camera LED via Linux (sysfs or similar).

Alternatively you could take the mailbox service that it uses and use it to control the LED yourself. (That should work against any GPIO, so you could adopt it as the standard way instead of using the GPIO library to wiggle GPIOs directly. That would make things consistent.)

waveform80, 2017-02-28:
Hmm, would you happen to know off the top of your head if the mailbox service is mapped into `/dev/gpiomem`? If not that'd require root which I'd like to avoid as a pre-requisite if at all possible.

6by9, 2017-03-01:
Mapped through /dev/vcio, not gpiomem. 660 permissions root:video. User pi appears to be a member of video by default.

https://github.com/6by9/rpi3-gpiovirtbuf/blob/master/rpi3-gpiovirtbuf.c is doing the same mailbox calls as the kernel driver and may be more readable. It doesn't require root on a standard Raspbian install.

waveform80, 2017-03-01:
Ah very handy! All the mailbox examples I've seen so far have been /dev/mem based (which is what I was hoping to avoid), but /dev/vcio looks like a perfect alternative.