waveform80/picamera #268: Capture continuous at multiple resolutions

jrosebr1, 2016-03-29:
I'm trying to utilize the `capture_continuous` function at multiple resolutions, with the end goal of obtaining each frame in NumPy format. I've gotten the following code to work for a single iteration:

<pre>from picamera.array import PiRGBArray
from picamera import PiCamera
import time

camera = PiCamera()
camera.resolution = (1024, 768)
camera.framerate = 30

highResCap = PiRGBArray(camera)
lowResCap = PiRGBArray(camera, size=(320, 240))

highResStream = camera.capture_continuous(highResCap, format="bgr",
	use_video_port=True)
lowResStream = camera.capture_continuous(lowResCap, format="bgr",
	splitter_port=2, resize=(320, 240))
time.sleep(2.0)
print("done warming up")

while True:
	lrs = lowResStream.next()
	lrFrame = lrs.array
	lowResCap.truncate(0)

	hrs = highResStream.next()
	hrFrame = hrs.array
	highResCap.truncate(0)

	print(lrFrame.shape)
	print(hrFrame.shape)</pre>

The code correctly prints out:

<pre>done warming up
(240, 320, 3)
(768, 1024, 3)</pre>

However, the script hangs after the first iteration. Eventually I have to ctrl+c out, which generates the following traceback:

<pre>  File "/home/pi/.virtualenvs/cv/local/lib/python2.7/site-packages/picamera/array.py", line 174, in bytes_to_rgb
    'Incorrect buffer length for resolution %dx%d' % (width, height))
picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 320x240</pre>

So it looks like I'm not properly truncating the stream.

I'm honestly not sure if I'm completely off base or if it's even possible to use `capture_continuous` in this way. If anyone can shed some light on this, I would really appreciate it.

jrosebr1, 2016-03-31:
@waveform80 Any suggestions here or am I totally off base? My goal is to re-create this code:

<pre>with picamera.PiCamera() as camera:
    camera.resolution = (1024, 768)
    camera.framerate = 30
    camera.start_recording('highres.h264')
    camera.start_recording('lowres.h264', splitter_port=2, resize=(320, 240))
    camera.wait_recording(30)
    camera.stop_recording(splitter_port=2)
    camera.stop_recording()</pre>

But using `capture_continuous`.

waveform80, 2016-05-07:
Hi Adrian - sorry for the long delay replying to this (and thanks for your excellent blog which I enjoy reading when I get the odd spare second!). I suspect in addition to truncate(0) in your original script you may need seek(0) as well. Incidentally you'd also need to specify `use_video_port=True` on the low res capture too otherwise it will default to the still port (with resulting mode switches) - it's not enough to just specify the splitter port.

Unfortunately I haven't got time to dig into this tonight - but I'll try and have a go tomorrow. Incidentally, in the forthcoming 1.11 you might want to have a look at the new [direct-capture-to-array abilities](http://picamera.readthedocs.io/en/latest/recipes2.html#capturing-to-a-numpy-array) - and there's some enhancements to RGB/BGR capture speed too (although there may also be some firmware upgrade pain which I'll need to detail in the docs).

jrosebr1, 2016-05-08:
Hey Dave, thanks for the reply. The next time I boot up my Pi, I'll be sure to try the `seek(0)` and `use_video_port=True` addition.

The direct-capture-to-array feature sounds really nice, I'm looking forward to giving it a try!

Zeesy, 2016-08-18:
Any resolution to this? To preface - why isn't it easier to just capture at a high resolution, and then create a separate "feed" or "stream" for downsampling and analysis? Is it because of the numpy arrays? I am working on the same problem right now, using `cv2.imshow` and `camera.capture(stream, 'rgb')` for my preview purposes. Stuck on the application of `seek(0)` - its current use as 

    rawCaptureLO.truncate(0)
    rawCaptureHI.truncate(0)
    rawCaptureLO.seek(0)
    rawCaptureHI.seek(0)
yields no different results from the same without `seek(0)` called. All my best attempts have resulted in the hi res stream frozen on the first frame captured. These include:

- a) initial rawCaptureHI frame in `camera.capture_continuous(format='bgr',splitter_port = 0, use_video_port=True)` consistently displayed alongside downsampled rawCaptureLO stream in `camera.capture_continuous(format='bgr',splitter_port = 1, use_video_port=True)`, with separate effect applied to rawCaptureLO
- b) initial rawCaptureHI frame consistently displayed alongside "blank" rawCaptureLO stream (not visible as video) that has been upscaled through conversion to a cv2 object and then `.resize`. Highly processor intensive.
- c)addition of `camera.add_overlay` displays a solid black fullscreen, with `cv2.imshow` windows behind regardless of specified `layer`. `PiOverlayRenderer` runs "Out of memory" after 15 triggered cycles, runs indefinitely when not triggered.
- d)addition of `camera.add_overlay` displays a flickering black fullscreen, with `cv2.imshow` windows behind regardless of specified `layer`. `PiOverlayRenderer` runs indefinitely when triggered, crashes with traceback to `use_video_port=True` within `camera.capture_continuous(rawCaptureLO, format='bgr', resize=(160,96),splitter_port=1, use_video_port=True)` when not triggered.

waveform80, 2016-08-19:
Unfortunately, no time to look into this currently - but some quick notes on c) and d):

c) the overlay (and the camera preview) will always appear on top of anything produced by the OS (including any X windows like OpenCV outputs). There's nothing I can do about this: see [FAQ#2](http://picamera.readthedocs.io/en/release-1.12/faq.html#can-i-put-the-preview-in-a-window)

d) See [FAQ#15](http://picamera.readthedocs.io/en/release-1.12/faq.html#preview-flickers-at-full-resolution-on-a-v2-module); the preview system won't handle full resolution on a V2 module (under any system; the same happens with raspistill if you use the -fp switch) so the preview (or any overlay) needs to run at a lower resolution

reillyardstard, 2016-09-05:
I just raised a bug related to a typo in picamera.array.bytes_to_rgb() i.e. (change fheith to fheight on line 110).  That will fix one issue you have here and things will run - but only for a short time (<1min for me) before falling over.  

Appears that bytes_to_rgb() also gets an empty data buffer occasionally which it doesn't like.  Fixing that in library-land was too hard for me, but it seems you can side-step it in app-land by just using the video-port for both resolutions.  Here's some working code:

```
from picamera.array import PiRGBArray
from picamera import PiCamera
import time
import cv2

camera = PiCamera()
camera.resolution = (1024, 768)
camera.framerate = 30

highResCap = PiRGBArray(camera)
lowResCap = PiRGBArray(camera, size=(320, 240))

highResStream = camera.capture_continuous(highResCap, format="bgr", use_video_port=True)
lowResStream = camera.capture_continuous(lowResCap, format="bgr", use_video_port=True, splitter_port=2, resize=(320, 240))
time.sleep(2.0)
print("done warming up")

while True:
    lrs = lowResStream.next()
    lrFrame = lrs.array
    lowResCap.truncate(0)

    hrs = highResStream.next()
    hrFrame = hrs.array
    highResCap.truncate(0)

    cv2.imshow('hires', hrFrame)
    cv2.imshow('lores', lrFrame)
    if cv2.waitKey(1) == 27:
        break
```

If you want to fix the alpha-stripping warning you get because of the splitting and resizing, then that's a different matter.  You can switch to bgra format but there's no 4-byte-per-pixel version of PiRGBArray to use.   Here's some code which will work pending the library getting it's own version:

```
import picamera.array
from picamera import PiCamera
from picamera.exc import PiCameraValueError
import time
import cv2
import numpy as np

class PiRGBAArray(picamera.array.PiRGBArray):
    ''' PiCamera module doesn't have 4 byte per pixel RGBA/BGRA version equivalent, so this inherits from the 3-bpp/RGBA/BGRA version to provide it
    '''
    def flush(self):
        self.array = self.bytes_to_rgba(self.getvalue(), self.size or self.camera.resolution)

    def bytes_to_rgba(self, data, resolution):
        ''' Converts a bytes objects containing RGBA/BGRA data to a `numpy`_ array.  i.e. this is the 4 byte per pixel version.
            It's here as a class method to keep things neat - the 3-byte-per-pixel version is a module function. i.e. picamera.array.bytes_to_rgb()
        '''
        width, height = resolution
        fwidth, fheight = picamera.array.raw_resolution(resolution)
        # Workaround: output from the video splitter is rounded to 16x16 instead
        # of 32x16 (but only for RGB, and only when a resizer is not used)
        bpp = 4
        if len(data) != (fwidth * fheight * bpp):
            fwidth, fheight = picamera.array.raw_resolution(resolution, splitter=True)
            if len(data) != (fwidth * fheight * bpp):
                raise PiCameraValueError('Incorrect buffer length for resolution %dx%d' % (width, height))
        # Crop to the actual resolution
        return np.frombuffer(data, dtype=np.uint8).reshape((fheight, fwidth, bpp))[:height, :width, :]

camera = PiCamera()
camera.resolution = (1024, 768)
camera.framerate = 30

highResCap = PiRGBAArray(camera)  # new 4-byte-per-pixel version below
lowResCap = PiRGBAArray(camera, size=(320, 240))

highResStream = camera.capture_continuous(highResCap, format="bgra", use_video_port=True)
lowResStream = camera.capture_continuous(lowResCap, format="bgra", use_video_port=True, splitter_port=2, resize=(320, 240))
time.sleep(2.0)
print("done warming up")

while True:
    lrs = lowResStream.next()
    lrFrame = lrs.array
    lowResCap.truncate(0)

    hrs = highResStream.next()
    hrFrame = hrs.array
    highResCap.truncate(0)

#    print(lrFrame.shape)
#    print(hrFrame.shape)
    cv2.imshow('hires', hrFrame)
    cv2.imshow('lores', lrFrame)
    if cv2.waitKey(1) == 27:
        break
```

PS Adrian, thanks for your blog - it's what inspired me to get into pi-python-picamera in the first place.
PPS Dave, thanks for picamera library - it's a great piece of work.

Zeesy, 2016-09-18:
Wow! Thanks Reilly, elegant solution to a problem that's been wracking my brain for months. .next is very useful.

Thanks Adrian and Dave as well, your blog and library (as well as all the tutorials and documentation) have been invaluable!

waveform80, 2017-02-02:
From reading the above I *think* this is fixed now (3a576ed2 fixes the typo @reillyardstard mentioned, the capture-to-buffer stuff landed in 1.11, and I'm reasonably confident the missing `seek(0)` was the solution to the original issue), so I'll close the ticket but do feel free to re-open it if you've any further queries regarding this!

closed 2017-02-02 by waveform80

mikevanis, 2017-03-07:
Hi guys, I'm trying to recreate the elegant solution suggested by @reillyardstard. Unfortunately, I'm getting @jrosebr1 's original error: `'Incorrect buffer length for resolution %dx%d' % (width, height))`. 

Any help would be much appreciated!

Here's my code below:

```
import cv2
import json
import time
import datetime
import os

# pi specific imports
from picamera.array import PiRGBArray
from picamera import PiCamera
import RPi.GPIO as GPIO

camera = PiCamera()
camera.resolution = (1024,768)
camera.framerate = 30
hiResCapture = PiRGBArray(camera)
lowResCapture = PiRGBArray(camera, size=(320,240))
hiResStream = camera.capture_continuous(hiResCapture, format="bgr", use_video_port=True)
lowResStream = camera.capture_continuous(lowResCapture, format="bgr", use_video_port=True, splitter_port=2, resize=(320,240))

time.sleep(config["camera_warmup"])

[...]

while True:
    # get new frame
    #image = frame.array
    lrs = lowResStream.next()
    image = lrs.array
    lowResCapture.truncate(0)
    lowResCapture.seek(0)

    hrs = hiResStream.next()
    hiresImage = hrs.array
    hiResCapture.truncate(0)
    hiResCapture.seek(0)
    # end of new frame

    cv2.imshow("Output", image)

    if GPIO.input(btnShutter) == False:
        takePhoto(hiresImage)
        time.sleep(0.5)

    key = cv2.waitKey(10)

```

mikevanis, 2017-03-15:
Hi @waveform80 - any ideas?

mikevanis, 2017-05-23:
Sorry to insist, but it would be fantastic to have a working example where two resolutions can be opened in order to have a hi res and a cv one.

abhisheksoni27, 2017-09-24:
I am facing the same issue, using picamera `v1.13`

abhisheksoni27, 2017-09-24:
Further, if you change this line https://github.com/waveform80/picamera/blob/master/picamera/array.py#L127, to this:

`'Incorrect buffer length %d for resolution %dx%d' % (len(data), width, height))`

This is what is returned:

```
picamera.exc.PiCameraValueError: Incorrect buffer length 1843200 for resolution 640x480
```

1843200 is exactly twice the size of the buffer the resolution requires. (`921600 = 640 * 480 * 3`)

mikevanis, 2017-09-25:
Thanks for the extra clude @abhisheksoni27. Have you had any further luck with it? Maybe this issue should be reopened? It doesn't look like `seek(0)` is fixing the problem.

ArMouReR, 2020-11-23:
Sorry for popping up so old thread, but I landed here trying to achieve the same as Adrian - simultaneous hi and low res stream. Is it possible in current version of the PiCamera ?