waveform80/picamera #269: Annotate for several splitters

cronnosli, 2016-04-04:
There is a way to insert a annotation per splitter?

Now I'm using the annotate_text as below
`self.camera.annotate_text = '{0} {2}'.format(self.banner, dt.datetime.now().strftime(self.dtFormat))`

But I wanna to insert a identification per video recorded.

6by9, 2016-04-05:
Not possible within the GPU code.
You'll note that the annotation is being set on ```self.camera```, so the camera component. That is common to all output streams. There is no similar support in any other component within the firmware.

You could look at writing the annotation yourself onto image buffers pulled back to the ARM.

waveform80, 2017-02-02:
Closing old tickets; do feel free to re-open if you've further queries about this!

closed 2017-02-02 by waveform80