waveform80/picamera #270: Several splitter_ports, video accelerated

cronnosli, 2016-04-05:
Hi, as my below code where I'm using multithread to record several videos(3), where any of then could start and stop at different moments, the picamera behave strange:
using just one splitter the video generate 60s in 60s OK
using two splitters each video generate 55s in 60s Not OK
using three splitters each video generate 38s in 60s Very Wrong

After the files are done I insert the header of the file eg:
`ffmpeg -i done-5050-2 -vcodec copy -f mp4 done-5050-2.mp4`

The file done-5050-2.mp4 should have a lenght of 60s but it is showing just 43s

`Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'done-5050-2.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf56.40.101
  Duration: 00:00:43.20, start: 0.000000, bitrate: 2881 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1440x1080, 2880 kb/s, 25 fps, 25 tbr, 1200k tbn, 2400k tbc (default)
    Metadata:
      handler_name    : VideoHandler`

What I'm doing wrong, or how should it work?

```python
import sys
import threading
import time
import datetime as dt
import picamera
import os
import os.path
from random import randint, random
from threading import Thread

class MoveCamChannel(threading.Thread):
	
	job = 0
	running = True
	quality = 20
	framerate = 60
	path = ''
	split = 60
	channel = 0
	timeout = 1800
	camera = None

	def __init__(self, job, camera, channel, path, framerate, split, quality, timeout):
		self.job = job
		self.channel = channel
		self.path = path
		self.framerate = framerate
		self.split = split
		self.quality = quality
		self.timeout = timeout
		self.camera = camera
		threading.Thread.__init__ (self)

	def stop(self):
		self.running = False

	def run(self):
		inter = 1
		file = '{0}-{1}'.format(self.job, inter)
		self.camera.start_recording('{0}/{1}'.format(self.path, file), format='h264', quality=self.quality, splitter_port = self.channel)
		start = dt.datetime.now()
		while self.running:
			nowTime = dt.datetime.now()
			diff = nowTime - start
			if diff.total_seconds() >= self.split:
				inter += 1
				fileNew = '{0}-{1}'.format(self.job, inter) 
				self.camera.split_recording('{0}/{1}'.format(self.path, fileNew), splitter_port = self.channel)
				self.move(file)
				file = fileNew
				start = dt.datetime.now()
			self.camera.wait_recording(1, splitter_port = self.channel)
		self.camera.stop_recording(splitter_port = self.channel)
		self.move(file)

	def move(self, file):
		if os.path.exists(file):
			os.rename('{0}/{1}'.format(self.path, file), '{0}/done-{1}'.format(self.path, file))

class MoveCamRecorder(threading.Thread):
	
	quality = 20
	framerate = 60
	path = ''
	width = 640
	height = 480
	split = 60
	led = 'off'
	fontSize = 20
	backgroundColor = 'black'
	banner = ''
	dtFormat = '%d/%m/%Y %H:%M:%S'
	timeout = 1800
	gravacao = {}
	camera = None

	def __init__(self, path, led, timeout):
		self.path = path
		self.led = led
		self.timeout = timeout
		threading.Thread.__init__ (self)

	def setBanner(self, fsize, fbg, banner, dtformat):
		self.fontSize = fsize
		self.backgroundColor = fbg
		self.banner = banner
		self.dtFormat = dtformat

	def setVideoSize(self, w, h):
		self.width = w
		self.height = h

	def setQuality(self, quality):
		self.quality = quality

	def setVideoSplit(self, split):
		self.split = split

	def setFramerate(self, fr):
		self.framerate = fr

	def startRecording(self, id):
		if len(self.gravacao) > 3:
			return False
		else:
			self.gravacao[id] = MoveCamChannel(
					id,
					self.camera,
					len(self.gravacao)+1,
					self.path, 
					self.framerate, 
					self.split, 
					self.quality, 
					self.timeout
			)
			self.gravacao[id].start()

	def stopRecording(self, id):
		if id in self.gravacao:
			self.gravacao[id].stop()
			self.gravacao[id].join()
			del self.gravacao[id]

	def detectCamera(self):
		try:
			with picamera.PiCamera() as camera:
				camera.start_recording('/dev/null', format='h264')
				camera.wait_recording(3)
			return True
		except Exception, e:
			self.recCamera = False
			return False

	def run(self):
		with picamera.PiCamera() as self.camera:
			self.camera.led = True if self.led == 'on' else False
			self.camera.resolution = (self.width, self.height)
			self.camera.framerate = self.framerate
			self.camera.start_preview()
			self.camera.annotate_background = picamera.Color(self.backgroundColor)
			self.camera.annotate_text_size = self.fontSize
			#self.camera.start_recording('/dev/null', format='h264', quality = 30, splitter_port = 0)
			start  = dt.datetime.now()
			while True:
				now = dt.datetime.now()
				diff = now - start
				self.camera.annotate_text = '{2} - {0} {1}'.format(self.banner, dt.datetime.now().strftime(self.dtFormat), diff.total_seconds())
				time.sleep(1)
			#self.camera.stop_recording(splitter_port = 0)

if __name__ == "__main__":
	recorder = MoveCamRecorder('./', 'on', 1800)
	recorder.setVideoSize(1440, 1080)
	recorder.setQuality(25)
	recorder.setFramerate(25)
	recorder.setVideoSplit(60)
	recorder.setBanner(40,'blue', 'Test', '%d/%m/%Y %H:%M:%S')
	recorder.detectCamera()
	recorder.start()
	time.sleep(5)
	recorder.startRecording('5050')
	time.sleep(20)
	recorder.startRecording('900')
	time.sleep(60)
	recorder.startRecording('1900')
	time.sleep(120)
	recorder.stopRecording('1900')
	time.sleep(30)
	recorder.stopRecording('900')
	time.sleep(5)
	recorder.stopRecording('5050')
```

6by9, 2016-04-06:
You're dropping frames and not looking at timestamps, so ffmpeg just glues together all the frames assuming 30fps.

Additional encodes are not free operations, and there are limits to the compute power.
The encoder is rated for a single stream at H264 Level 4.1, which is the equivalent of approx 1080P30 (technically it is 62,914,560 luma samples/s). You can get a little bit more than that out (1080P33 was always the aim), and you can split that encode power between multiple streams with a small overhead.

If I'm reading your code correctly then you're pushing 1440x1080@25fps through, so 38,880,000 luma samples/s. Two streams would be 77,760,000 luma samples/s and you're beyond the hardware spec and it will drop frames.

I'm also not sure which encoding formats PiCamera is setting up the splitter ports with. I suspect it is having to convert a load of images between planar and an internal image format which is going to hit memory bandwidth.

cronnosli, 2016-04-06:
I was able to recording several videos at same time using a encoder class and passing it to an object. Than the object use the frame buffer to write in several ffmpeg pipes, using it I have several videos in h264 mp4 without loss frames or using too much harware from raspi.

Thanks for your explanation.

waveform80, 2017-02-02:
Looks like this ticket's answered so I'll close it (do feel free to re-open if you've further queries regarding this).

closed 2017-02-02 by waveform80