waveform80/picamera #271: Capture frame by frame with H264

CamAuPoint, 2016-04-06:
Hi,
I bought the Multiplexer IV PORT (link : https://github.com/ivmech/ivport) to have 4 Picamera on my Raspberry Pi 2, and I am trying to record video from the 4 cameras at the same time.

Right now, I'm able to take picture at the same time (with 0.007 second delay) in 4 seperates folder (CAM1, CAM2, CAM3 and CAM4). The delay it's for the switch from the 1rst cam to the second cam with the GPIO switch "cam_change()"

Here's the script that take JPG from all the cameras

`

    import time
    import picamera
    import subprocess
    import RPi.GPIO as gp

    gp.setwarnings(False)
    gp.setmode(gp.BOARD)

    gp.setup(7, gp.OUT)
    gp.setup(11, gp.OUT)
    gp.setup(12, gp.OUT)

    gp.output(7, False)
    gp.output(11, False)
    gp.output(12, True)

    frames = 200

    cam = 1

    def cam_change():
      global cam
      gp.setmode(gp.BOARD)
        if cam == 1:
           # CAM 1 for A Jumper Setting
           gp.output(7, False)
           gp.output(11, False)
           gp.output(12, True)

        elif cam == 2:
           # CAM 2 for A Jumper Setting
           gp.output(7, True)
           gp.output(11, False)
           gp.output(12, True)

        elif cam == 3:
           # CAM 3 for A Jumper Setting
           gp.output(7, False)
           gp.output(11, True)
           gp.output(12, False)

        elif cam == 4:
           # CAM 4 for A Jumper Setting
           gp.output(7, True)
           gp.output(11, True)
           gp.output(12, False)

        cam += 1
        if cam > 4:
           cam = 1

        def filenames():
            frame = 0
            while frame < frames:
            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            cam_change()        # Switching Camera
            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            yield '/home/pi/camaupoint/camaupointssd/TEST/cam1/image%02d.jpg' % frame
            frame += 1

            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            cam_change()        # Switching Camera
            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            yield '/home/pi/camaupoint/camaupointssd/TEST/cam2/image%02d.jpg' % frame
            frame += 1

            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            cam_change()        # Switching Camera
            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            yield '/home/pi/camaupoint/camaupointssd/TEST/cam3/image%02d.jpg' % frame
            frame += 1

            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            cam_change()        # Switching Camera
            time.sleep(0.007)   # SD Card Bandwidth Correction Delay
            yield '/home/pi/camaupoint/camaupointssd/TEST/cam4/image%02d.jpg' % frame
            frame += 1

      with picamera.PiCamera() as camera:
           camera.resolution = (2592, 1944)
           camera.rotation = 180
           camera.framerate = 5
           camera.start_preview()

      time.sleep(2)    # Camera Initialize
      start = time.time()
      camera.capture_sequence(filenames(), use_video_port=True)
      finish = time.time()

      print('Captured %d frames at total %.2ffps' % (frames, frames / (finish - start)))
`

The script is working fine, it takes 200 frames, and each folder (CAM1, CAM2, CAM3 and CAM4) have 50 JPG from his cam. My question is, Instead of taking JPG pictures, how can I record each camera in H264 with the best FPS possible ?

Like on that video (For example) : https://www.youtube.com/watch?v=w4JZN7Y0d2o

At the beggining of the YouTube video, we see that we are video capturing 2 cameras at the same time, how can I acheived this with H264 ?

waveform80, 2016-04-06:
I'd assume they're not recording video from their two cameras but doing rapid image capture while switching between cameras then stitching the results together afterward. Knowing a little about what the switcher does behind the scenes I'd be surprised if it was possible to record video and switch cameras without breaking the recording (I vaguely recall the firmware devs mentioning that they were rather amused the product worked at all - something about it relying on the firmware being resilient to camera resets which was only added for some crazy customer way back when!).

Anyway, I'm afraid I can't offer much in the way of support here - it's not my product and I've no experience of using one, but if anyone more in the know wants to chip-in feel free!

CamAuPoint, 2016-04-06:
Hmmm, that's a good point. Maybe they are stitching the results together like you say. But with PiCamera is it possible to manipulate each frame in a stream ?

If I'm recording into a stream, and after I take the
 - 1rst frame and put it into CAM1.h264
 - take the 2nd frame and put it into CAM2.h264
 - take the 3rd frame and put in into CAM3.H264
 - take the 4th frame and put it into CAM4.h264
 - take the 5th frame and put in into CAM1.h264
 - take the 6th frame and put in into CAM2.h264
ect...

Can I acheived this with a python script ? What is the possibilities ?

Thanks in advance !

6by9, 2016-04-06:
I've not played with one, but the above description is how I believe the switcher works - it's using an error recovery mechanism. The only other possibility is that it is doing something very weird on the I2C to allow only one sensor (I2C slave) to respond at a time. The switching time is impressively fast if it is just the error recovery mechanism.

You can't split up an H264 stream in the way you suggest.
H264 generates motion vectors for the differences between successive frames (called P-frames for predicted), so you MUST have the previous frame in order to be able to reconstruct the current one. You could make all frame I-frames (don't reference any other frames), but then the compression ratio sucks. Or you could look at MJPEG which has no form of temporal compression, but again the compression ratio is significantly worse.

CamAuPoint, 2016-04-06:
Oh ok ! But is it possible to tell (with a python script) that the previous frame was 4 frame ago ?