waveform80/picamera #272: Dev dependency on lsb-release and libjpeg8-dev

eldog, 2016-04-07:
Following the development installation instructions in the documentation it appears that `make develop` fails due to `lsb-release` and `libjpeg8-dev` not being installed.

I think the system dependency installation command should be

$ sudo apt-get install build-essential git git-core exuberant-ctags \
    libjpeg8-dev lsb-release python-virtualenv

waveform80, 2016-04-07:
Good catch :) I'll sling this into the instructions for 1.11

eldog, 2016-04-07:
Cool : ) Also I think `python-dev` is required too.

waveform80, 2016-04-07:
Hmm ... might be, but that's generally only needed for C-based extensions and picamera's pure python (well, sort of)

eldog, 2016-04-07:
I think it caused `make develop` to crash when `pip` tried to install Pillow.

waveform80, 2016-04-08:
Ah, of course - that's a dep of the test suite. Yup, that'll need to go in too.

commit 9fc812f6 (2016-04-17): Fix #272

Re-include the development instructions (someone actually read them!)
and get the dependencies right...
closed 2016-04-17 by waveform80