waveform80/picamera #273: Is it possible to have more than one previews on screen?

VV-Cephai, 2016-04-10:
I'm trying to showa preview-window and next to it another zoomed in preview-window from the same camera. Is this at all possible?
I tried to create a seperate instance of PiPreviewRenderer, but I can't connect it to the camerea so far.

6by9, 2016-04-12:
It's possible via MMAL, but I don't know how easy it would be to wrap into PiCamera.

The zoom will have to be done by setting the display rectangles appropriately on the renderers - the camera component does not support different zoom/crop on different ports. crop (http://picamera.readthedocs.org/en/release-1.10/api_renderers.html#picamera.renderers.PiRenderer.crop) looks to be the relevant parameter on the renderer.

Whether you do it as 
```camera_preview_port -> render1
camera_video_port ->render2```
or
```
camera_preview_port -> splitter-> render1 
                               -> render2
```
is almost irrelevant if you can use MMAL_ENCODING_OPAQUE between them all, and assuming you don't have to tear too much apart to use the video port.

waveform80, 2017-02-02:
As @6by9 suggests it's not possible with the high level picamera API. However, the lower level mmalobj API that picamera is built on can indeed do this. A very similar question recently cropped up [on the RPi stack exchange](http://raspberrypi.stackexchange.com/questions/60119/split-screen-cloned-display/60126#60126) and I threw together a little demo script for it:

```python
from picamera import mmalobj as mo, mmal
from signal import pause

camera = mo.MMALCamera()
splitter = mo.MMALSplitter()
render_l = mo.MMALRenderer()
render_r = mo.MMALRenderer()

camera.outputs[0].framesize = (960, 720)
camera.outputs[0].framerate = 30
camera.outputs[0].commit()

p = render_l.inputs[0].params[mmal.MMAL_PARAMETER_DISPLAYREGION]
p.set = mmal.MMAL_DISPLAY_SET_FULLSCREEN | mmal.MMAL_DISPLAY_SET_DEST_RECT
p.fullscreen = False
p.dest_rect = mmal.MMAL_RECT_T(0, 180, 960, 720)
render_l.inputs[0].params[mmal.MMAL_PARAMETER_DISPLAYREGION] = p
p.dest_rect = mmal.MMAL_RECT_T(960, 180, 960, 720)
render_r.inputs[0].params[mmal.MMAL_PARAMETER_DISPLAYREGION] = p

splitter.connect(camera.outputs[0])
render_l.connect(splitter.outputs[0])
render_r.connect(splitter.outputs[1])
pause()
```

(copied from the link above)

That should give you a decent starting point for this (sorry it's taken so long to reply - I'm gradually working through old tickets!)